opentx/opentx #2232: Feature Request: Model based Flight-Battery Class

wolkstein, 2015-04-26:
Hi,
a Model based flight Battery Class would be a cool feature.

Try to explain the idea and why it would be usefully.
I fly opentx in combination with ardupilot-copter(pixhawk) in a professional setup. We have a Areal Video, Foto and Mapping service.
If you fly for example on film sets you often have to fly only one scene and than you have to wait to your next action. this is sometimes only a few minutes but also sometimes a longer time.  in this time you mostly power of the copter and also the transmitter. one of the main problems here is that you loose all battery informations like flight-time, consumption ma/h and in my case also consumed Wh . if you restart the copter and the transmitter every of this interesting battery value is lost on your custom telemetry screen. of course you can see the battery voltage. but only reading the battery voltage do not give you an good idea how many power is left in your flight battery.
To fix this problem A model based battery class which save battery status to sd card would the goal.

I am create two Ideas:
1)
  A simple singleton class which only holds some battery informations from the last flight. In this case a user can for example activate USE_Battery in in Model Menu. And also if use battery is set in model menu users can define some basic Model battery infos.(capacity in mah and wh, min capacity left in %, minimum cell voltage , critical and warning cell voltage and so on??. )  now if USE_Battery is set the user get a question at powering the transmitter. for example NEW Battery. if you disagree this question the battery class restore all battery data from last flight which was saved on sd-card. in lua script users can call the battery class. for example battery.getConsumption(). and also he can store battery data for example by calling battery.setConsumption( int ). so users can customise there telemetry screens with some simple battery class functions.

APM battery consumption example:
apm set battery consumption at powering the system to zero. if you get mavlink based battery consumption by using a mavlink to s-port converter you are able to display this value on telemetry screen.
```lua
local batteryConsumptionOfsett = 0

-- in init function
batteryConsumptionOfsett = battery.getConsumption() -- 0 if a new battery is in use

-- in main loop we will display real battery consumption 
--value 218 for example holds apm mavlink based battery consumption
lcd.drawNumber(10,10, batteryConsumptionOfsett + getValue(218) ,MIDSIZE)
-- storing the new value to battery class for next fly
battery.setConsumption( batteryConsumptionOfsett + getValue(218)  )
’’’
also users can create custom functions by calling other battery class data. for example an model based mah(fuel) gauge which now can get informations like mah of battery and min power left by calling battery.getInfoMaH() and  battery.getInfoMinPowerLeft()

-------

2)
More complex Battery Class implementation based on the first idea but with the possibility to create N Model Based Batteries.
Here users can create N batteries for each model. And users can set more individual Battery informations.
In this case you can choose one of your batteries if transmitter is powered. and also you are able to charge this batteries by choosing charge battery(N) or all. in this case the class also can hold and store more relevant data to sd card. like battery cycles, livetime. also voltage drop and so on. so it would be possible to monitor your battery health. also you can see  on a flight day which batteries are in which condition. 
--------

/g
wolke

kilrah, 2015-04-27:
mAh and timers can already be stored at power off, and in 2.1 this will be extended to all telemetry values.
In 2.1 lua scripts will also have the ability to save/read data from the SD card, so IMO anything you may need is already covered.

wolkstein, 2015-04-27:
it is also possible to catch and release events successfully in telemetry scripts? to create for example an popupInput on your custom telemetry script? also it would helpfully to execute some code on destroy. for example to use I/O model based functions(sounds that they are supported in 2.1) to store custom data to sd card at transmitter shutdown.

/g
wolke

bsongis, 2015-04-27:
popupInput in a telemetry screen? What do you have in mind as a script?

wolkstein, 2015-04-28:
if I try to implement my flight-battery class, as described on top, only with lua scripts than i need an popup input(each time i start my taranis) where I can select an battery, and also where i can select to charge a desired battery or charge all. some days ago i try this to implement. but i fail on different problems. first problem was, that a popup in telemetry screen do not work as an popup in one time scripts. at least i noticed that key-events will not released after catching them. second problem was that i can not use the lua I/O Model in scripts(2.0.15). so i found out that an flight-battery class as i have it in mind is not realistic project only with lua here. that was the reason why i create this Feature Request for such an Flight Battery Class. IMO an real professional feature what help people to increase safe flying. and also monitor batteries live-time on taranis is one of the best choices of all.

reasons:
 * my taranis telemetry is the one which run 99% save. it mostly never fail or run out of range against for example my second telemetry the 3dr-radio which lost connection mostly after some meters:)
* the taranis is the main TOOL i use in my flight setup. i also use an professional flight data log tool on android to create my flight logs(RPAS Logger). but here i have to input all battery log data manually. this need a lot of time. also the battery function on RPAS logger is only based on flight timer. this timer you must also start and stop by clicking on android screen.

if you fly professional jobs you need simple and solid tools which you can put inside your pre and post flight check-list procedure. that why i think the taranis is the best place for such a battery monitoring feature.

/g
wolke

closed 2020-07-08 by stale[bot]