opentx/opentx #2239: PPM modulation opendrive/fulldrive

djdevil, 2015-04-29:
Hello i update my new 9xrPro to last version of opentx but tne my module assan v1 not work!! The problem is the protocol of ppm modulation..where i can change modulation ppp fulldrive?there is no menu!! Thx

projectkk2glider, 2015-04-29:
What do you mean by *fulldrive*? You can change polarity and timings on the first page of model setup.

djdevil, 2015-04-29:
In the menu of ersky9r firmware  In the Protcol Menu there is an option to Set PPM Drive to FullDrive and opedrive!

Full Drive outputs the PPM signal driving both high and low. OPENDRAIN only drives it low, expecting a pullup resistor, in the module, to pull it high. This was added because at least one module type caused a problem by passing the PPM voltage back through the module power pin preventing correct detection of power being switched off.

djdevil, 2015-05-01:
any Reply???

projectkk2glider, 2015-05-02:
It is clear now what you want.

djdevil, 2015-05-02:
Thx i hope this fuction for the next update

projectkk2glider, 2015-05-02:
I am not sure we have a spare location in eeprom for this new setting. @bsongis will know more. Maybe we could split `ModuleData::ppmPulsePol`?
```c
PACK(typedef struct t_ModuleData {
  int8_t  rfProtocol;
  uint8_t channelsStart;
  int8_t  channelsCount; // 0=8 channels
  uint8_t failsafeMode;
  int16_t failsafeChannels[NUM_CHNOUT];
  int8_t  ppmDelay;
  int8_t  ppmFrameLength;
  uint8_t ppmPulsePol;
}) ModuleData;
```

djdevil, 2015-05-02:
Please...i want opentx on my radio

bsongis, 2015-05-03:
@projectkk2glider Yes we could change the ModuleData structure (but only in 'next'). Perhaps we could even save one or two bytes in the whole structure

djdevil, 2015-05-03:
Thank you very much😀

bsongis, 2015-05-03:
@projectkk2glider Will you take care of this issue? At least the EEPROM change should be done before 2.1 is out to avoid problems ...

projectkk2glider, 2015-05-03:
Ok, I will try.

closed 2015-05-04 by bsongis

projectkk2glider, 2015-05-04:
@djdevil This is now done, but only in the upcoming OpenTX 2.1 version which is not yet airworthy! Use with caution.

PR #2255 (2015-05-04): Projectkk2glider/issue 2239 ppm pushpull  for 9xrpro
This is an attempt to fix #2239.

@bsongis I want you to tell me if I am going in the right direction. I've first tried to add opendrain option to the existing line for PPM parameters (length, pause, polarity) but there isn't enough space. I find working with menus very hard! The MENU_TAB is very hard to read/edit.

When this is completed I will rebase and squash commits to make it into one commit.