opentx/opentx #2241: Increase number of logical switches

Matty2B, 2015-05-01:
A fairly self explanatory one this... is there sufficient system resources/space within the EEPROM to have a greater number of logical switches? I have never run out until recently, but for my Dad's Spitfire we have a lot of complexity going on and I have maxed out all 32. By no means an urgent one though, I appreciate this will matter only to a very limited number of power users.

projectkk2glider, 2015-05-05:
Maybe your particular problem could be better solved with a Lua script. Lets face it. Having 32 LS used gets pretty messy and hard to maintain.

I any case there is always compromise between the number of some features and used space. With current model storage scheme (in EEPROM) there is not a lot of maneuvering space.

bsongis, 2015-05-07:
Right 32 switches seem to me a max on this radio with the resources we have.

closed 2015-05-07 by bsongis