opentx/opentx #2247: Jenkins build scripts

xmxwx, 2015-05-02:
Where can I find scripts used for building plans on [OpenTX's Jenkins service](http://jenkins.open-tx.org:8080/)?

Before posting a pull request I try building the project after my changes using my own script. However, it would be much better if I've done it exactly the same way Jenkins does. Recently I broke Jenkins build because my script didn't include compiling a simulator build. Of course, CI service like Jenkins is there to catch such mistakes, but if the scripts were available it could save us a bit of hassle.

kilrah, 2015-05-02:
cd radio/src/

# OpenTX on 9X stock
make clean
make PCB=9X EXT=FRSKY HELI=YES
make simu PCB=9X EXT=FRSKY HELI=YES 
make gtests EXT=FRSKY HELI=YES
./gtests

# OpenTX on Taranis Plus
make clean
make PCB=TARANIS PCBREV=REVPLUS HELI=YES LUA=YES
make simu PCB=TARANIS PCBREV=REVPLUS HELI=YES LUA=YES
make gtests PCB=TARANIS PCBREV=REVPLUS HELI=YES LUA=YES
./gtests

# OpenTX on Taranis
make clean
make PCB=TARANIS HELI=YES LUA=YES
make simu PCB=TARANIS HELI=YES LUA=YES
make gtests PCB=TARANIS HELI=YES LUA=YES
./gtests

# OpenTX on Sky9x
make clean
make PCB=SKY9X EXT=FRSKY HELI=YES
make simu PCB=SKY9X EXT=FRSKY HELI=YES
make gtests PCB=SKY9X EXT=FRSKY HELI=YES
./gtests

# OpenTX on Gruvin9x
make clean
make PCB=GRUVIN9X EXT=FRSKY SDCARD=YES
make simu PCB=GRUVIN9X EXT=FRSKY SDCARD=YES

# OpenTX on MEGA2560
make clean
make PCB=MEGA2560

# Companion
cd ../../buildir/
make

xmxwx, 2015-05-02:
Thank you very much!

xmxwx, 2015-05-02:
BTW. In order to get gtests compiling on my system (Ubuntu 14.04 Trusty Tahr) I had to add -I/usr/include/qt4 -fPIC to gtests' Makefile rule. It currently seems to be impossible to pass these options to make without modifying Makefile. Add another issue for that?

bsongis, 2015-05-03:
I am really afraid of somebody adding a rm * in a commit which would break everything ...

xmxwx, 2015-05-03:
@bsongis 
> I am really afraid of somebody adding a rm * in a commit which would break everything ...

What do you mean by that? You don't want to have the Jenkins build scripts in the repository in order to protect that installation from someone malicious?

@kilrah, you've posted no `make gtests` on PCB=MEGA2560. Is it on purpose? When I tried `make gtests PCB=MEGA2560` it failed on a lot of Mixer tests:

    [==========] 55 tests from 10 test cases ran. (10850 ms total)
    [  PASSED  ] 43 tests.
    [  FAILED  ] 12 tests, listed below:
    [  FAILED  ] Mixer.R2029Comment
    [  FAILED  ] Mixer.Cascaded3Channels
    [  FAILED  ] Mixer.CascadedOrderedChannels
    [  FAILED  ] Mixer.Cascaded5Channels
    [  FAILED  ] Mixer.SlowOnSwitch
    [  FAILED  ] Mixer.SlowUpOnSwitch
    [  FAILED  ] Mixer.SlowOnSwitchAndPhase
    [  FAILED  ] Mixer.SlowOnSwitchSource
    [  FAILED  ] Mixer.DelayOnSwitch
    [  FAILED  ] Mixer.SlowOnSwitchReplace
    [  FAILED  ] Mixer.NoTrimOnInactiveMix
    [  FAILED  ] getSwitch.recursiveSW
    
    12 FAILED TESTS

xmxwx, 2015-05-03:
I posted PCB=MEGA2560 gtests failures as a separate issue: #2250

griebd, 2015-05-05:
I have searched for and haven't found the build options used in the "official build server" of OpenTX firmware (the firmware companion downloads for us).

I think it should be available somewhere so people who wants to do custom builds can start from a know working build, the official one. (that is why I'm looking for those options, in my case, taranis plus)

The build serve uses exactly those same options from Jenkins test server? (plus the options marked on companion)

Thanks!

xmxwx, 2015-05-05:
What @kilrah posted was enough for my needs, so I didn't reopen this issue. But now, @griebd has similar concerns to mine and it may turn out that more people seem to need such information. I think it would be good reopen this issue and close it when it is clear for everyone how to perform builds just like on the official build server...

kilrah, 2015-05-06:
It's a bit of a mess currently. Some defaults come from the makefile and some from the compilation server script. Would need a bit of a cleanup (all standard build options in makefile).

kilrah, 2015-05-06:
For the Taranis we're nearly there though.

> make PCB=TARANIS PCBREV=REVPLUS HELI=YES GVARS=YES LUA=NO_MODEL_SCRIPTS 

should give you the standard build.

xmxwx, 2015-05-08:
> BTW. In order to get gtests compiling on my system (Ubuntu 14.04 Trusty Tahr) I had to add -I/usr/include/qt4 -fPIC to gtests' Makefile rule. It currently seems to be impossible to pass these options to make without modifying Makefile. Add another issue for that?

I've just commited a very simple solution (ce58a11e73b47114f12804821f88d1f71abe4f59) to this problem - a new Makefile variable: SIMUCPPFLAGS.

xmxwx, 2015-05-11:
So... getting back to the topic...

Do you know who has got full administrative rights to OpenTx's Jenkins and can send us (or maybe commit in the repo) the scripts which are used for the build plans?

To contribute something from myself, here's the script I've recently knit (based on info from this issue) and started using before committing and creating pull requests:

    #!/bin/bash
    set -e
    
    export SIMUCPPFLAGS="-I/usr/include/qt4 -fPIC" # Needed on my Ubuntu 14.04
    
    compile_and_run_tests() {
            echo
            echo
            echo "=============================================================================================================="
            echo "  $*"
            echo "=============================================================================================================="
            echo
            make clean
            make "$@"
            make simu "$@"
            make gtests "$@"
            ./gtests
    }
    
    # Answer to my question on github
    # https://github.com/opentx/opentx/issues/2247#event-295520726
    
    # OpenTX on 9X stock
    compile_and_run_tests PCB=9X EXT=FRSKY HELI=YES
    
    # OpenTX on Taranis Plus
    compile_and_run_tests PCB=TARANIS PCBREV=REVPLUS HELI=YES GVARS=YES LUA=NO_MODEL_SCRIPTS
    
    # OpenTX on Taranis
    compile_and_run_tests PCB=TARANIS HELI=YES LUA=YES
    
    # OpenTX on Sky9x
    compile_and_run_tests PCB=SKY9X EXT=FRSKY HELI=YES
    
    # OpenTX on Gruvin9x
    compile_and_run_tests PCB=GRUVIN9X EXT=FRSKY SDCARD=YES
    
    # OpenTX on MEGA2560
    compile_and_run_tests PCB=MEGA2560
    
    # a few additional combinations
    compile_and_run_tests PCB=9XR128 TELEMETRY_MOD=14051_SWAPPED EXT=FRSKY
    compile_and_run_tests PCB=9XR128 TELEMETRY_MOD=14051 EXT=FRSKY
    compile_and_run_tests PCB=9XR128 EXT=FRSKY
    compile_and_run_tests PCB=9X2561
    compile_and_run_tests PCB=9XRPRO

kilrah, 2015-05-12:
There is no "build plan", but a php script that builds on demand based on the options in companion. The firmware string (default downloaded firmware filename) is what decides what's built.

Jenkins is only used for the nightlies and commit check builds, and those are the ones supposed to catch errors.

I suggested having the script in the repo at some point, not sure why it wasn't added. Might be because there are personal details in (email addresses for failure notifications) and it was too much hassle removing them but re-adding them when putting the scripts back etc.

xmxwx, 2015-05-17:
I think that settings like the e-mail addresses for failure notifications could be moved to a separate configuration file. This way each deployment of the script could have its own settings and there would be no need to change it on upgrades. Of course except for cases when the configuration variables themselves are changed or new ones are introduced.

commit 72652e42 (2015-12-03): Fixes #2247: Initial state of Jenkins build scripts
closed 2015-12-03 by projectkk2glider