opentx/opentx #2248: As of 2.0.15, setting the volume with a special function is not sticky + SF cannot set volume from GV6

marcmerlin, 2015-05-02:
I tried to have a special function to set the volume, but only if a switch was down. This worked.
But as soon as the switch is back up, volume gets reset to whatever the default is.

I ended up doing something pretty complicated to get this to work (too many hours for me to admit):
1) L23: differential of S2 > 1 and SA down
2) SF21: If L23, play value GV6
3) SF22: if L3, Adjust GV6 from Source S2
4) SF23: ON Volume CH18

I really would have liked Volume to take GV6 as an input, but it won't, so I had to take it from CH18
5) Mix Ch18: MAX Weight(GV6)

In other words, I use Ch18 only to copy the GV6 value so that SF23 can use that value to set volume

Things to fix
1) Volume should be set-able from a GV value
2) Somehow it seems that it's a bug that volume gets reset by something in the background and the only way to keep it to another level is to have a SF to constantly reset it.

kilrah, 2015-05-03:
AFAIK 2) is the way it has always worked. The Volume SF simply overrides the system set volume when active.

marcmerlin, 2015-05-03:
@kilrah thanks for confirming I didn't do anything wrong or hit a weird bug.
May I suggest that this doesn't make sense and isn't welcome? If I'm setting volume from a SF, I don't expect it to be reset behind my back to another value.
In my case, I didn't want S2 to be used just for volume, but only to set it when a certain switch is set, so that I can use it for other things otherwise.

kilrah, 2015-05-03:
> May I suggest that this doesn't make sense and isn't welcome?

Well not so much as it's been that way for about 4 years and you're the first to complain ;)

Anyway you found the solution to it if you want it done differently, use a channel.
Leaving that open for discussion about including GVARs.

marcmerlin, 2015-05-03:
@kilrah, ok, I get your point :)
So, what we have:
1) this is not a big or urgent problem since I'm the first person to notice and report it, as you say.
2) decide if this is expected and wanted behaviour, or only a byproduct of how the code works today
3) Allow to set volume from a GVAR since volume must be stored somewhere to be reset constantly if it can't be read from a pot

Anyway, thanks for reading and considering.

fraternl, 2015-05-13:
Are there plans to change the way you can change the master volume?
I like it that it can be set in a custom way. I, for instance, am using a curve so it is impossible to completely mute it. This to prevent a situation where no warnings can be heard.

It now has to be programmed into each model and intuitively it belongs more to some global setting. I believe there's already a ticket for that. I understand this poses some problem on who is owning a certain control. 
If a priority is given to "the model" that problem might be solved?

kilrah, 2015-05-13:
2.1 has radio-wide special functions indeed. No curves though...

marcmerlin, 2015-12-29:
Does the new opentx allow setting volume from a GV now?
Having to copy my GV to a channel and then set the volume from a channel value is a bit weird as a workaround :)

kilrah, 2015-12-30:
No.

bsongis, 2015-12-30:
GVars redesign explained in #3185

closed 2015-12-30 by bsongis