opentx/opentx #2249: Wrong ADC channels cosen on MEGA2560

olterion, 2015-05-02:
In the file /targets/common_avr/adc_driver.cpp is the line 54: ADCSRB &= ~(1<<MUX5);
This line changes the ADCSRB register from initial MUX5=1 to MUX5=0. So the ADC channels 0-7 are choosen. In opentx.cpp line line 1503 is the call of "adcPrepareBandgap" for the MEGA2560 board too.  In the program "getADC_bandgap" the ADCSRB value will be changed back to MUX5=1 only for GRUVIN9X board. So the MEGA2560 board has a bug, because the ADC cannels 0-7 will be read. 
I solved this problem for me with comment out the line 54 in adc_driver.cpp, but this will not work for every board.

JivaroFAD, 2015-05-03:
Status: Fixed

bsongis, 2015-05-03:
I am lost with this issue, it's fixed now, right? Please update the milestone as well!

xmxwx, 2015-05-03:
Oh, this bug seems to be [introduced by me while adding adc_driver.cpp](https://github.com/opentx/opentx/commit/485edb579d63f1f5025e397f0e407c04fd336ff9#commitcomment-11016031). Sorry!

xmxwx, 2015-05-03:
Wouldn't it be better if we replaced

    #if defined(CPUM2560) && !defined(PCBMEGA2560)

back with the original?

    #if defined(PCBGRUVIN9X)

...and merged ADCSRB and ADMUX back together? Otherwise, now on PCBMEGA2560 we pointlessly manipulate ADMUX because we're going to ignore the measured voltage reference value and [take fixed value instead](https://github.com/opentx/opentx/blob/next/radio/src/targets/common_avr/adc_driver.cpp#L99).

bsongis, 2015-05-03:
Yes your natural fix is better, would you commit it to 'next' and close the related issue? Thank!

xmxwx, 2015-05-03:
Done.

closed 2015-05-03 by xmxwx

bsongis, 2015-05-03:
Thanks a lot!

xmxwx, 2015-05-03:
No problem. It's a shame I've introduced this bug in a first place...

bsongis, 2015-05-03:
We all do mistakes!!! Thanks for your help