opentx/opentx #2250: PCB=MEGA2560 gtests failures

xmxwx, 2015-05-03:
make clean && make PCB=MEGA2560 && make gtests PCB=MEGA2560 && ./gtests

...first failures:

    [ RUN      ] Mixer.R2029Comment
    tests/mixer.cpp:471: Failure
    Value of: (1024*256)
      Actual: 262144
    Expected: chans[1]
    Which is: 0
    tests/mixer.cpp:475: Failure
    Value of: 0
    Expected: chans[1]
    Which is: 262144
    tests/mixer.cpp:479: Failure
    Value of: (1024*256)
      Actual: 262144
    Expected: chans[1]
    Which is: 0
    [  FAILED  ] Mixer.R2029Comment (1 ms)

...summary:

    [==========] 55 tests from 10 test cases ran. (10568 ms total)
    [  PASSED  ] 43 tests.
    [  FAILED  ] 12 tests, listed below:
    [  FAILED  ] Mixer.R2029Comment
    [  FAILED  ] Mixer.Cascaded3Channels
    [  FAILED  ] Mixer.CascadedOrderedChannels
    [  FAILED  ] Mixer.Cascaded5Channels
    [  FAILED  ] Mixer.SlowOnSwitch
    [  FAILED  ] Mixer.SlowUpOnSwitch
    [  FAILED  ] Mixer.SlowOnSwitchAndPhase
    [  FAILED  ] Mixer.SlowOnSwitchSource
    [  FAILED  ] Mixer.DelayOnSwitch
    [  FAILED  ] Mixer.SlowOnSwitchReplace
    [  FAILED  ] Mixer.NoTrimOnInactiveMix
    [  FAILED  ] getSwitch.recursiveSW

    12 FAILED TESTS

projectkk2glider, 2015-05-03:
This works, no problems here (on tag 2.0.16):
```
make clean
make PCB=MEGA2560 EXT=FRSKY HELI=YES
make PCB=MEGA2560 EXT=FRSKY HELI=YES gtests
./gtests 
```

Output:
```
$ ./gtests 
[==========] Running 58 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 7 tests from Timers
[ RUN      ] Timers.timerReset
[       OK ] Timers.timerReset (0 ms)
[ RUN      ] Timers.saveRestoreTimers
[       OK ] Timers.saveRestoreTimers (0 ms)
[ RUN      ] Timers.timerOff
[       OK ] Timers.timerOff (0 ms)
[ RUN      ] Timers.timerAbsolute
Timer[0] negative
Timer[0] stopped state at 260
[       OK ] Timers.timerAbsolute (2 ms)
[ RUN      ] Timers.timerThrottle
Timer[0] negative
Timer[0] stopped state at 260
[       OK ] Timers.timerThrottle (1 ms)
[ RUN      ] Timers.timerThrottleRelative
Timer[0] negative
Timer[0] stopped state at 260
[       OK ] Timers.timerThrottleRelative (3 ms)
[ RUN      ] Timers.timerThrottleTriggered
Timer[0] THr triggered
Timer[0] THr triggered
Timer[0] negative
Timer[0] stopped state at 260
[       OK ] Timers.timerThrottleTriggered (2 ms)
[----------] 7 tests from Timers (9 ms total)

[----------] 5 tests from EEPROM
[ RUN      ] EEPROM.100_random_writes
[       OK ] EEPROM.100_random_writes (8902 ms)
[ RUN      ] EEPROM.test2
[       OK ] EEPROM.test2 (371 ms)
[ RUN      ] EEPROM.eeCheckImmediately
eeprom write model
[       OK ] EEPROM.eeCheckImmediately (401 ms)
[ RUN      ] EEPROM.copy
[       OK ] EEPROM.copy (460 ms)
[ RUN      ] EEPROM.rm
[       OK ] EEPROM.rm (372 ms)
[----------] 5 tests from EEPROM (10506 ms total)

[----------] 10 tests from Trims
[ RUN      ] Trims.throttleTrim
[       OK ] Trims.throttleTrim (0 ms)
[ RUN      ] Trims.invertedThrottlePlusThrottleTrim
[       OK ] Trims.invertedThrottlePlusThrottleTrim (0 ms)
[ RUN      ] Trims.throttleTrimWithZeroWeightOnThrottle
[       OK ] Trims.throttleTrimWithZeroWeightOnThrottle (0 ms)
[ RUN      ] Trims.invertedThrottlePlusthrottleTrimWithZeroWeightOnThrottle
[       OK ] Trims.invertedThrottlePlusthrottleTrimWithZeroWeightOnThrottle (0 ms)
[ RUN      ] Trims.greaterTrimLink
[       OK ] Trims.greaterTrimLink (0 ms)
[ RUN      ] Trims.chainedTrims
[       OK ] Trims.chainedTrims (0 ms)
[ RUN      ] Trims.infiniteChainedTrims
[       OK ] Trims.infiniteChainedTrims (0 ms)
[ RUN      ] Trims.CopyTrimsToOffset
[... 85 lines cut ...]
[ RUN      ] Lcd.Prec2_Left
[       OK ] Lcd.Prec2_Left (1 ms)
[ RUN      ] Lcd.Prec2_Right
[       OK ] Lcd.Prec2_Right (1 ms)
[ RUN      ] Lcd.Line_Wrap
[       OK ] Lcd.Line_Wrap (1 ms)
[ RUN      ] Lcd.vline
[       OK ] Lcd.vline (1 ms)
[----------] 6 tests from Lcd (6 ms total)

[----------] 4 tests from getSwitch
[ RUN      ] getSwitch.undefCSW
[       OK ] getSwitch.undefCSW (0 ms)
[ RUN      ] getSwitch.circularCSW
[       OK ] getSwitch.circularCSW (0 ms)
[ RUN      ] getSwitch.nullSW
[       OK ] getSwitch.nullSW (0 ms)
[ RUN      ] getSwitch.recursiveSW
[       OK ] getSwitch.recursiveSW (0 ms)
[----------] 4 tests from getSwitch (0 ms total)

[----------] Global test environment tear-down
[==========] 58 tests from 12 test cases ran. (10532 ms total)
[  PASSED  ] 58 tests.
```

bsongis, 2015-05-03:
Perhaps the problem comes in 'next'?

projectkk2glider, 2015-05-03:
I will check that now.

projectkk2glider, 2015-05-03:
Yes, the problem exists on `next` branch only. @xmxwx please specify such important detail in future.

projectkk2glider, 2015-05-03:
One reason for this is different implementation of function `switchState()`in master https://github.com/opentx/opentx/blob/master/radio/src/targets/mega2560/board_mega2560.cpp#L124 and next https://github.com/opentx/opentx/blob/next/radio/src/targets/mega2560/board_mega2560.cpp#L109

But which one is right? @JivaroFAD did the inversion of switches in next in 55a7fbaa2e0bfb68234ca8e5a22085c7340a5f95

JivaroFAD, 2015-05-03:
I did the inversion to have, with pull-up on and switches connected to ground : 
- nothing connected on M2560 input => input @ 5V => default state of switch = 0
- input put to ground => input @ 0V => state of switch = 1

Allthought all seems to work fine at beginning of marh 2015, I found last week a similar bug with EXT=FRSKY and HELI=no (flash compiled but radio doesn't run), that disapear with HELI=yes. I didn't make for this moment, on radio or with gtests.

xmxwx, 2015-05-07:
> I did the inversion to have, with pull-up on and switches connected to ground

@JivaroFAD: but the very first schematic you put into the repo (in 0306734) already had switches shorting to the ground. So before 55a7fba the code was inconsistent with the schematics?

If so... then we just need to modify the simu target to follow up to your changes.

closed 2015-05-07 by xmxwx

JivaroFAD, 2015-05-07:
Thanks :)
There was not contradiction between code and schematic before 55a7fba, but there was two constraints : all unused inputs had to be connected to ground to have switches at 0 state on OpenTX and switches activation was at opening. 
Now, all inputs have 0 state by default in software if not connected to a switch, it is much simpler and natural.

xmxwx, 2015-05-07:
You're welcome :-)

commit 7755bf03 (2015-05-07): Merge pull request #2270 from xmxwx/next

Fixes #2250
commit d63f6bca (2015-05-07): Fixes #2250
commit aa7cf0ec (2015-05-10): Fixes #2250