opentx/opentx #2253: Companion 2.0.16 crashing

Alexander-C, 2015-05-03:
I am running Win 8.1 and version 2.0.16 of Companion, crashes whenever 'Simulate Tx' or 'simulate' is pressed.  Never happened on previous version.

projectkk2glider, 2015-05-03:
Hmm, just downloaded official Companion 2.0.16 and installed on Windows XP 32bit (in virtualbox) and it works perfectly. 

@Alexander-C do you have any more information, any output of crash?

Alexander-C, 2015-05-03:
[image]

I am using Windows 8.1 64bit. This problem has only occured since installing 2.0.16

bsongis, 2015-05-03:
Reproduced here. 'master' on Windows 7.1

kilrah, 2015-05-03:
Can't reproduce on neither my Win7 PC or Win8.1 tablet with the official release build...

bsongis, 2015-05-03:
I removed the contents of my Companion directory and the problem is solved!
I don't know which file was involved, would you try to do the same, but keeping a backup of your old directory so that you can find which file it is?

kilrah, 2015-05-03:
Ah I can reproduce if I enable UAC!

kilrah, 2015-05-03:
Yes it must be the stdout.txt! That should only be enabled in debug builds!

bsongis, 2015-05-03:
Yes I had this file in my directory when I removed the contents

kilrah, 2015-05-03:
OK, something between 2.0.15 and 2.0.16 has caused companion to try and create stderr.txt and stdout.txt files in the program directory on launch, but I can't find out what change causes that. 
I tried commenting out @projectkk2glider 's  fprintf's to stderr for SDL init errors in case QT decides to add that by itself it it sees such a thing but it doesn't seem to solve the problem. 
On Win7 and up writes to the Program Files directory are forbidden, which causes the crash.

I have UAC disabled on my machines so that goes unnoticed. In the meantime people can run companion as admin...

griebd, 2015-05-03:
ok, haven't look at any code at all, but when I use companion simulator in my linux box I have lots of output to my terminal, like bellow...
```
f_opendir(/home/user/taranis_sd//SOUNDS/en/SYSTEM) = OK
findTrueFileName(/home/user/taranis_sd//SCRIPTS/WIZARD/wizard.lua)
        found: wizard.lua
f_stat(/home/user/taranis_sd//SCRIPTS/WIZARD/wizard.lua) = OK
findTrueFileName(wizard.lua)
        found: wizard.lua
f_open(wizard.lua) = 0x7f06e401a0c0
f_close(0x7f06e401a0c0)
eeprom write model
f_opendir(/home/user/taranis_sd//SOUNDS/en/MODEL01) = error 2 (No such file or directory)
playFile("/SOUNDS/en/SYSTEM/thralert.wav", flags=0, id=0)
findTrueFileName(/home/user/taranis_sd//SOUNDS/en/SYSTEM/thralert.wav)
        found: thralert.wav
f_open(/home/user/taranis_sd//SOUNDS/en/SYSTEM/thralert.wav) = 0x7f06ec012b00
findTrueFileName(plane.bmp)
        found: plane.bmp
f_open(plane.bmp) = 0x7f06e4025950
f_close(0x7f06e4025950)
[...]
```

bsongis, 2015-05-03:
debug.cpp:

These 2 lines could be removed when DEBUG not defined ...

fputs(tmp, stdout);
fflush(stdout);

projectkk2glider, 2015-05-03:
Careful, we now have a stdout and stderr redirection in Companion to capture that output and show it in a Debug window (F6 in simulator)!

projectkk2glider, 2015-05-03:
Its a SDL artifact: http://sdl.beuc.net/sdl.wiki/FAQ_Console

projectkk2glider, 2015-05-03:
I think the problems is that we use main() from SDL http://stackoverflow.com/questions/15564016/does-anyone-know-how-to-eliminate-creation-of-stdout-txt-and-stderr-txt-when-sdl (second answer) when linking Companion.

projectkk2glider, 2015-05-03:
Ever since I made audio in companion simulator, the windows build failed http://jenkins.open-tx.org:8080/job/OpenTX%20Stable%20-%20Nightly%20builds/266/console. That is until Bertrand fixed cmake to use SDL winmain 345d980bb172f3e65a356672418c2bc72faa49d6. Could we use QT winmain as before?

bsongis, 2015-05-03:
Right I applied the same trick on simulator than the one which was used in companion! I am ok to change this of course, just need to have something compiling on both Linux and Windows ... and Apple

projectkk2glider, 2015-05-03:
Hold on, the SDL winmain is actually needed, because it initializes some stuff for SDL.

I have found that if one defines environment variable `SDL_STDIO_REDIRECT` to `0`, then those two files are not created! This is from SDL source:
```c
	/* Check for stdio redirect settings and do the redirection */
	if ((env_str = SDL_getenv("SDL_STDIO_REDIRECT"))) {
		if (SDL_atoi(env_str)) {
			redirect_output();
		}
	}
#ifndef NO_STDIO_REDIRECT
	else {
		redirect_output();
	}
#endif

```

projectkk2glider, 2015-05-03:
But I am not sure why the heck redirect_output() is called if SDL_STDIO_REDIRECT is not defined!?

bsongis, 2015-05-03:
Ok I search as well in this direction

bsongis, 2015-05-03:
Their use of NO_STDIO_REDIRECT is really strange! We have to define this redirect environment variable, no choice!

projectkk2glider, 2015-05-03:
Yes, after I defined it on my system it behaves as it should. No more stdout.txt and stderr.txt files.

bsongis, 2015-05-03:
Yes, but this is not an acceptable solution for the average user! I guess we have to define this variable at runtime on startup.

projectkk2glider, 2015-05-03:
Can't be done in Companion itself, when we get to main() it is already too late.

bsongis, 2015-05-03:
Then we cannot use this main() :(

projectkk2glider, 2015-05-03:
The flow is:
* windows calls winMain() - this one is in SDL library. There is also one in QT library, but that is not as good for SDL stuff (so I've read).
* the winMain() does some stuff and then calls our main()

Same flow for both winMain() functions.

bsongis, 2015-05-03:
Yes, but if we call winMain from SDL we have this issue (stdout.txt and stderr.txt created). I would prefer we revert to the winMain from Qt!

projectkk2glider, 2015-05-03:
Try it, if joystick and simu audio still work, then its good. 

This is the source of SDL 1.2 winMain(). I don't see anything special inthere:
```c
/* This is where execution begins [windowed apps] */
#ifdef _WIN32_WCE
int WINAPI WinMain(HINSTANCE hInst, HINSTANCE hPrev, LPWSTR szCmdLine, int sw)
#else
int WINAPI WinMain(HINSTANCE hInst, HINSTANCE hPrev, LPSTR szCmdLine, int sw)
#endif
{
	HMODULE handle;
	char **argv;
	int argc;
	char *cmdline;
	char *env_str;
#ifdef _WIN32_WCE
	wchar_t *bufp;
	int nLen;
#else
	char *bufp;
	size_t nLen;
#endif

	/* Start up DDHELP.EXE before opening any files, so DDHELP doesn't
	   keep them open.  This is a hack.. hopefully it will be fixed 
	   someday.  DDHELP.EXE starts up the first time DDRAW.DLL is loaded.
	 */
	handle = LoadLibrary(TEXT("DDRAW.DLL"));
	if ( handle != NULL ) {
		FreeLibrary(handle);
	}

	/* Check for stdio redirect settings and do the redirection */
	if ((env_str = SDL_getenv("SDL_STDIO_REDIRECT"))) {
		if (SDL_atoi(env_str)) {
			redirect_output();
		}
	}
#ifndef NO_STDIO_REDIRECT
	else {
		redirect_output();
	}
#endif

#ifdef _WIN32_WCE
	nLen = wcslen(szCmdLine)+128+1;
	bufp = SDL_stack_alloc(wchar_t, nLen*2);
	wcscpy (bufp, TEXT("\""));
	GetModuleFileName(NULL, bufp+1, 128-3);
	wcscpy (bufp+wcslen(bufp), TEXT("\" "));
	wcsncpy(bufp+wcslen(bufp), szCmdLine,nLen-wcslen(bufp));
	nLen = wcslen(bufp)+1;
	cmdline = SDL_stack_alloc(char, nLen);
	if ( cmdline == NULL ) {
		return OutOfMemory();
	}
	WideCharToMultiByte(CP_ACP, 0, bufp, -1, cmdline, nLen, NULL, NULL);
#else
	/* Grab the command line */
	bufp = GetCommandLine();
	nLen = SDL_strlen(bufp)+1;
	cmdline = SDL_stack_alloc(char, nLen);
	if ( cmdline == NULL ) {
		return OutOfMemory();
	}
	SDL_strlcpy(cmdline, bufp, nLen);
#endif

	/* Parse it into argv and argc */
	argc = ParseCommandLine(cmdline, NULL);
	argv = SDL_stack_alloc(char*, argc+1);
	if ( argv == NULL ) {
		return OutOfMemory();
	}
	ParseCommandLine(cmdline, argv);

	/* Run the main program (after a little SDL initialization) */
	console_main(argc, argv);

	/* Hush little compiler, don't you cry... */
	return 0;
}
```

bsongis, 2015-05-03:
I get again the old issue, both companion and simulator don't want to link, WinMain is defined twice. Searching for a tweak ...

bsongis, 2015-05-03:
The #undef main trick seems working ... checking further ...

bsongis, 2015-05-04:
Is it possible someone else double check the fix:
http://jenkins.open-tx.org/nightly-20/2015_05_04/

closed 2015-05-04 by bsongis

Alexander-C, 2015-05-04:
Tried this build and seems to work okay.

jermoo, 2015-05-04:
I had an issue with the companion 2.0.16 upgrade crashing. I manually uninstalled and re-installed. It works. It may be something related to windows/upgrade rather than the actual code.

projectkk2glider, 2015-05-04:
@jermoo did you try http://jenkins.open-tx.org/nightly-20/2015_05_04/companionInstall_2.0.17.exe, describe your system .

On my virtual machine (32bit Windows XP) it works perfectly, the two files are not created.

Alexander-C, 2015-05-04:
I tested version 2.0.16 not 2.0.17 so cannot comment

projectkk2glider, 2015-05-04:
You must test the one I posted, otherwise the test is pointless. @Alexander-C if 2.0.16 is working for you then the .17 should also and your system is not affected by this it seems.

projectkk2glider, 2015-05-04:
Maybe I am wrong. Just checked the times and perhaps the 2.0.16 in that url also contains the fix.

Alexander-C, 2015-05-04:
I only tested 2.0.16 as that was the release version and the problem seems to be fixed.

commit cb13ad1e (2015-05-03): Fixes #2253
commit 1609b123 (2015-05-04): Fixes #2253