opentx/opentx #2257: LCD refactorization proposal: alignment

xmxwx, 2015-05-04:
Currently, `lcd_outdezNAtt()` by default starts printing the number from right and advances leftwards. The LEFT attribute changes this behavior by making it calculate the expected width of the number to be printed, pre-advance (rightwards) the starting position and then proceed with leftward printing. All other LCD-related functions ignore the LEFT attribute.

My proposal is to provide bidirectional printing, i.e. to have e.g. RIGHT_TO_LEFT attribute which would cause `lcdPutPattern()` to print each character leftwards; this would remove the need to calculate the expected width of the text to be printed and thus greatly simplify the code in some places. A natural expected effect of such a refactor would be that all LCD-printing functions would then behave consistently and - as a bonus - we'll have the ability to right-align the text (e.g. for textual telemetry values).

What do you think about that?

I've got a proof of concept stashed somewhere on my clone of the master branch. But since it's quite a huge change, I'll need to rebase it onto the next branch, thoroughly test it (hence my recent interest in gtests) and post it here for a review.

bsongis, 2015-05-04:
Yes I do agree, this change is nice. LEFT could be the default behavior. But only on TARANIS! I am really afraid it would increase flash usage on AVR boards.

xmxwx, 2015-05-05:
> But only on TARANIS! I am really afraid it would increase flash usage on AVR boards.

It would be of no use to me since currently I only own 9XR. I am definitely targetting AVR boards :-)

If I managed to do it without increasing flash usage, wouldn't you have anything else against introducing it on AVR? If didn't manage to do it without increasing flash usage, how big increase would you accept?

BTW Doing such a refactor I'd like to be sure that nothing gets silently broken (fortunately, there are already many gtests implemented, maybe it would be good to implement a few more on this occasion), but also that it doesn't degrade the performance. A gross error would be easily noticed, but small degradations of performance in subsequent code changes could go unnoticed until someone pays closer attention to it. How do you estimate performance of the code? Static analysis of the source code? Profiling on the simulator? Profiling on the target boards?

bsongis, 2015-05-05:
I would like we don't add anything to flash usage, we are already really short! But if it fits, yes it will be  a nice change. Take care on AVR boards, the LcdFlags is only 8bits, we reuse flags, take care there will be no conflict with another flag!

xmxwx, 2015-08-05:
> Take care on AVR boards, the LcdFlags is only 8bits, we reuse flags, take care there will be no conflict with another flag!

From what I see in the code already a lot of effort was put into squeezing all the LCD options into 8-bit flag word. This looks quite difficult, but not impossible.

CONDENSED is marked "TODO remove on TARANIS". It seems to be used only:
 1. in lcd_outhex4() for getting A-F letters of the same width as 0-9 digits
 2. in putsFlightMode() implementation to print out flight mode without "FM" prefix, however this in turn doesn't seem to be used anywhere.

On ARM, CONDENSED could be replaced by simply using SMLSIZE font. For AVR to save FLASH space, we could still produce it for A-F letters on-the-fly from STDSIZE exactly the same way it is done for CONDENSED. Or alternatively, maybe FIXEDWIDTH would be enough? This would need checking.

Moreover, BOLD flag is allocated like it was a general option, while in fact it is just another font face. Both on ARM and AVR it is impossible to have BOLD on any other FONTSIZE than the standard one.

On AVR, putting BOLD, DBLSIZE and optionally SMLSIZE together in the same subfield of lcdflags will use two bits instead of the current three (incl. CONDENSED). On ARM, also one bit would be saved and the could would be "a bit" ;-) cleaner.

closed 2020-07-08 by stale[bot]