opentx/opentx #2258: 2.0.17 Mixermon

MIKE-DAILEY, 2015-05-04:
Mixermon is working on my Taranis Plus fine, but is not working in Companion simulator.  Not sure if I'm doing something wrong. 

Mike

bsongis, 2015-05-05:
In 2.0.x the Companion simulator uses only standard options, not the options the user chooses.

closed 2015-05-05 by bsongis

kilrah, 2015-05-05:
There's no reason not to include it in the simu though, is there?

LapinFou, 2015-05-05:
I think there is a very good reason. I guess most of the pilots will develop their complex program under Companion. Having the MIXERS MONITOR would be very nice in order to debug complex mathematical operations. Specially when the mix result is going over 100% !
An example to illustrate how useful it is: with the "mixers monitor" you will see and understand why the channel (with a 'slow' parameter) that you are programming takes some time to move. You will see that you max channel is going over 100% and reach 200%. Since you got a "slow" parameter, your channel must recover from +200% to +100% before you actually see any movement on the "servos monitor".
Without the "mixer monitor", it can be quite difficult to understand and correct the problem.
So, I think, this option should also be implemented in Companion.
As I said, I'm always programming my models from my PC. So, to me, this option is even more useful under Companion as in the Taranis itself.

kilrah, 2015-05-05:
I said no reason NOT to include it. And it's in now as you can see with the 2 commits.

LapinFou, 2015-05-05:
Oups !!! Sorry kilrah, I misread your comment.
My apologize.
Will you release soon the 2.0.18, in order to be align with the firmware?

pindeslandes, 2015-08-05:
I know this is closed but wouldn’t it be more convenient if Companion could show the mixers monitor in some tabs here, as for outputs?
[image]

LapinFou, 2015-08-05:
Yes !!! I love this idea !
:+1:

projectkk2glider, 2015-08-05:
Create a new issue or this will get forgotten.

pindeslandes, 2015-08-06:
Done!

mpaperno, 2017-02-06:
This seems to have gotten lost somewhere along the way, `MIXERS_MONITOR` is no longer defined by default for Taranis simu/libsimulator builds.  We could force it in simu/CMakeLists but of course then there's no way to build simu w/out it... but maybe that doesn't matter?

kilrah, 2017-02-06:
The defaults are not in the code, but in the release build scripts on the server. Before 2.2.0 release we'll have to do a little rundown of the useful options to include by default for each variant. Since it changes during development we don't bother doing that before it's final.

mpaperno, 2017-02-06:
Sounds good, yes, best to handle it in the build scripts with the rest of the defaults.

commit 27bba73d (2015-05-05): Mixers monitor in simulator (Fix #2258)
commit f4c3610a (2015-05-05): Mixers monitor in simulator (Fix #2258)