opentx/opentx #2262: Taranis automatic shutdown on low battery

hillflyer, 2015-05-05:
I accidently deep discharged the Taranis battery because I did not hear the "Battery critical" warnings.
Since the shutdown is controlled by the firmware and not directly by the power switch, my wish would be for an automatic shutdown feature in addition to the existing acoustic warnings.

jtaylor2, 2015-05-05:
That would be extremely dangerous.  Imagine the results of shutting down the transmitter while someone is flying a 40 pound plane (or even a small foamy for that matter).  Even doing it after a period of inactivity would be dangerous because it could still be controlling a plane someone was working on.

Jim

bsongis, 2015-05-05:
Anyhow it is impossible to force a shutdown on Taranis!

closed 2015-05-05 by bsongis