opentx/opentx #2263: Display text (on LCD) from telemetry stream

zdar, 2015-05-05:
There is a patch progress for cleanflight which will send text over telemetry back to trasmitter. The patch was primarility trgeted for the HOTT however it could be extended for other platforms. I can imagine that it could be implemented in a way to work with FrSky+OpenTX. 

Display of the text stream is very useful fo several scenarios but let me mention three major ones:
- status messages
- alerts
- text terminal (for configuration and larger texts)
The biggest advantage here is that you move from LCD on model to the transmitter.

Status message
I'd like to have the possibility to have some kind status line on the main screen of transmitter LCD with text from model controller via telemetry. Example - flight mode name, armed/disarmed...
It would be much more useful than currently displayed model name (at least for me).

Alerts
A full transitter LCD screen message = An addition for current alerts but triggered by text message from telemetry stream and displaying message coming from model.

Text terminal
There is in progress/pending pull request/patch which will allow cleanfight to display configuration menus on transmitter via telemetry text stream (=terminal screen on trasmitter). It would be nice to extend OpenTX telemetry screens to include "teminal" screen for telemetry telemetry too. It would open a whole lot of new opportunities

--------

I have looked into FrSky telemetry (HUB) protocol and it could be extended to transport at least low bandwith text (somebody knows the baudrate of telemetry?). It would need to add tree custom IDs (for each text stream type) messages and transmit the text char by char.

Is there interest and possibility to implement such thing on OpenTX ? I can implement thing on the other side however my knowledge of OpenTX limits to user role...

xmxwx, 2015-05-05:
I have quite similar objectives. Recently, I've hacked on displaying ArduPilot flight mode as another telemetry variable. The proof of concept was successful. However, to include it into the official opentx it needs some more work (e.g. #2257, but more issues probably to come).

kilrah, 2015-05-06:
The current intention is to allow access to the raw telemetry data stream in Lua. You could then write whatever UI you'd like as a lua script, and communicate bidirectionally.

Text is too limiting, with what we have in mind the whole MSP protocol could likely be implemented.

I'd have liked to have it in 2.1, but we're a bit in a hurry with it for other reasons so unless someone joins the effort that might have to wait a bit.

zdar, 2015-05-06:
@kilrah  The general approach is nice but there are difficulties at least for my setup (9XR+FrSky)

For practical reasons my telemetry is one way only which I belive is a case for a lot of people. That means there is no way to get data from transmitter to the receiver/control board. This greatly reduces MSP usefulness (but does no rule MSP out).

The biggest challenge is available telemetry data rate and data loss (remeber there is probably no way to tell receiver to retransmit lost packets). If I have googled well the available data rate is around 100-200 chars per second. Which means it is good for occasional messages but nothing intensive.

LUA - is LUA supported on atmel? It requires SD card which is not available for my platform (please correct me if I'm wrong).

kilrah, 2015-05-07:
> my setup (9XR+FrSky)

Any planned improvements will indeed only apply to Taranis platform and later, and using smart port equipment.

zdar, 2015-05-07:
Ok. Do I understood well that for "old" setup as mine no such feature is planned to supported?

---

I was thinking about having small text buffer (one line or maybe two - at least for status bar) in telemetry structure which would be filled from telemetry via Hub/D protocol custom ID. My initial idea is to use first byte as address (0 reserved for control codes)  in the buffer and the second one for the char itself. This would make the buffer less prone to the corruption due to the packet loss and only chars would be needed to be transmitted.

If I hack something together is there chance to get it back to opentx?

kilrah, 2015-05-07:
> Do I understood well that for "old" setup as mine no such feature is planned to supported?

Not by the "usual" OpenTX developers at least. Nobody in the team uses AVR platforms anymore, and we already don't have time to do all we'd want on what we use...

> If I hack something together is there chance to get it back to opentx? 

Yes sure. As long as it's clean enough and doesn't break anything either in OpenTX or in terms of compatibility with "genuine" sensors of course ;) 
Depending on the flash memory use it might have to be implemented only for atmega128 and not for 64.

zdar, 2015-05-09:
I have created first proof of concept. It implements kind of status line under model name (model name uses normal text to give some space).

I have in this commit 
https://github.com/zdar/opentx/commit/e785a9ff64c7edaa74abdb1f7846f890acae34eb

Is something like this acceptable?

I have it working with cleanflight and displaying armed sttatus + flight modes.

xmxwx, 2015-05-09:
The idea seems fine to me. However:
 1. I don't see the code which would fill in FrskyData::text[]. Did you forget to add your changes in radio/src/telemetry/frsky*.cpp to the commit?
 2. Maybe it would be nice to allocate field IDs with FrSky, so that ideas like this one (but from different people, different projects) don't collide in the FrSky Hub protocol?
 3. Don't change irrelevant (whitespace-only) changes in the commit. The whitespaces sure need to be corrected, but it would be better done in a single commit cleaning up whole repository.
 4. In the source code please keep the comments English-only. :-)
 5. Development of new features should be rather done on the 'next' branch.

zdar, 2015-05-09:
@xmxwx 
1. Sorry missed to commit one file https://github.com/zdar/opentx/commit/e18f0eee04728f47ee73429c8a71915f366a0772

2. There is a problem with OpenTX and FrSky IDs. For some reason OpenTX code does not use ID codes inside its code but it uses ID directly as position in data struct. That means if I use higher ID I have to increase size of the struct (and waste some more memory). It was probably meant to simplify things however it causes problems (and wastes memory) now and it probably should be rewritten.

3.+4. I will take care about this when making final patch :)

5. Ok. I will prepare new patch for the next branch. There is some work needed to be done on the code before it is ready for public testing. For example from initial tests I see that some kind of transmition error check needs to be implemented.

----

Memory question - how much memory could be consumed on AVR platforms? I would like to try to implement full telemetry text screen. That means that I need memory for bigger buffer (8*24 = 192 -> 200 bytes)?

kilrah, 2015-05-09:
> There is a problem with OpenTX and FrSky IDs. For some reason OpenTX code does not use ID codes inside its code but it uses ID directly as position in data struct.

This is the kind of optimizations that were needed simply to actually get those things to run on the AVR in the first place. Using the struct offset directly as an index removes the need for a big fat switch() that would otherwise be needed to put things at the right place and would use a ton of program memory we really don't have much of. If you look well no RAM is wasted as each gap in the struct caused by unused IDs has been reused to store other information, only 5 bits are actually unused in the whole struct.

> Memory question - how much memory could be consumed on AVR platforms? I would like to try to implement full telemetry text screen. That means that I need memory for bigger buffer (8*24 = 192 -> 200 bytes)?

It's pretty much impossible to predict accurately. But if I build what was my standard FW for the 9X it uses 3500 bytes of the 4K (static). If you add the stack to that and consider that other sets of options may use more, I would guesstimate no more than 50-100 bytes.

closed 2019-08-19 by kilrah