opentx/opentx #2265: Profiling/benchmarking the code on the target platform

xmxwx, 2015-05-05:
I plan to propose a code refactor (particularily #2257, but it applies to any bigger refactor), however, I'd like to be sure that the changes I made are not only correct but also don't introduce any performance degradation.

I think it would be nice to have a possibility to benchmark some parts of the code on the target platform. I think about adding an optional compilation parameter which would enable instrumentation code which would use CPU timer to measure how long it takes to execute some routines. The measurements could be performed once on power-on, manually triggered via menu or remote triggered via serial port, with results displayed on LCD, sent through serial port or stored on EEPROM to be read via bootloader or avrdude.

What do you think?

projectkk2glider, 2015-05-06:
I think I remember having something like this in one of my branches at home. It is a light weight implementation of timers much like the existing timers for mixers duration for example. I will post my implementation later.

xmxwx, 2015-05-17:
@projectkk2glider did you manage to find some of your code which would help to achieve this?

projectkk2glider, 2015-05-17:
Sorry, forgot about this. 

Take a look at commit 328439525f2b3c6b3be2cf7466a2db96cb5bef15 from my branch https://github.com/projectkk2glider/opentx/tree/projectkk2glider/issue1849_delay_measuring

It contains a class DebugTimer which can measure the time of code execution. For short times (less than 30ms) it uses high precision timer and for higher times a low precision system tick timer.

xmxwx, 2015-05-30:
Thanks! I've used your code, extended it to support AVR platform, increased debug_timer_t resolution to 10 us (on AVR it exceeds the 64 us resolution of the internal ~16 kHz timer) and used it to measure the fix for #2319. If #2322 gets accepted, I think this issue could be closed as well. If not, we can cherry-pick only the DebugTimer-related stuff.

closed 2019-11-15 by stale[bot]