opentx/opentx #2266: Feature Request-One click setting of Failsafe settings

billmars, 2015-05-05:
I saw in "[Enhancement] Increase Failsafe security #2117" that you were discussing setting Failsafe options.  Setting the failsafes are obviously a really smart idea but my personal challenge with them is knowing what to set for each channel.  There is no way of knowing how much up elevator is enough to slow the decent or fly level without having the airplane dive or climb and stall, etc.  I was thinking a great way to address this would be to use the momentary switch in the same manner as the instant trim function to automatically set the failsafe settings for all channels instantly.  This way you could take off, climb to a safe altitude and configure the airplane in its landing mode (ie gear down, flaps down, proper amount of throttle and elevator to ensure a slight decent or level flight, etc).  Once the airplane is behaving as you want it to, flicking the momentary switch would then "save" those settings to the failsafe settings for that model.  That would be a very quick and accurate way of making sure all channels are correct and nothing is missed.  Thoughts?

bsongis, 2015-05-08:
Good idea.

kilrah, 2015-05-08:
Pretty sure that one was already refused in the past... Sounds good BUT very prone to error IMO, it would be easy to trigger a "Set Failsafe" SF my mistake even just when removing it after you're done...

bsongis, 2015-05-08:
Right, it will be included in the range of "Dangerous functions" such as ModuleOff / RangeCheck / ModelBind

ckleanth, 2015-05-08:
bear in mind this valid only for  X type RX. I dont think there is any way for this to work for the D type (or type of RX) and as such can be sure this function is tested against these senario?
thanks

kilrah, 2015-05-08:
In D8 mode none of the failsafe settings do anything, there isn't even a way to communicate them.

commit 377e3ee3 (2015-05-11): Fixes #2266
SetFailsafe Custom Function
commit 4729edb7 (2015-05-11): Fixes #2266
SetFailsafe Custom Function
commit 998bcf52 (2015-06-28): Fixes #2266
SetFailsafe Custom Function
commit 55356bae (2015-06-28): Fixes #2266 - SetFailsafe Custom Function. Version without an EEPROM
shift. It removes the ModuleOff function which nobody will need I hope
...
closed 2015-06-29 by bsongis