opentx/opentx #2267: Log Viewer 2.0.17

flarssen, 2015-05-06:
1. It's not possible to show more than two graphs simultaneously. When #3 is selected, only #1 is displayed.   

2. Selecting a row in the table view makes X and Y scales jump to unreal ranges.

griebd, 2015-05-06:
1.yes, it's possible to select as many as you want... if the ranges are too far way, they are plotted all in the say Yaxis! Is that what you are saying? if not, please elaborate so I can try to help...

If your problem is that with all plots in the same Yaxis you get only lines, that was already questioned and will be enhanced with multiple Yaxis.

2. selecting one row wasn't something I expected someone to do... will look at it and, if it is possible, will make the plot correct, if not, will plot nothing until more rows are selected.

flarssen, 2015-05-06:
Log example is at the bottom of this post: http://www.rcgroups.com/forums/showpost.php?p=31380873&postcount=1
Right, it seems that the problem is that it switches to single Y scale if more than two is selected. I didn't initially notice the current and vfas graphs in the RPM scale, so I assumed they were gone completely.
Expected to get more scales as need, but I guess that's what already has been suggested.

griebd, 2015-05-06:
For item 2: now the scales make sense.
BUT the lib used (QCustomPlot) simple doesn't plot a single point. So, although the values in the axis are as expected, there is no plotting at all for a single selected row.

romoloman, 2015-05-06:
Well about multiple axis, i got a different idea, multiple axis can be good up to a limited number of plots, but with many plots it get messy (i tried it).  
The idea could be to scale each plot to the maximum and display the axis only when we select a plot.

griebd, 2015-05-07:
actually there won't be a axis for each plot, only when ranges are too far way, like RPM and CELL, a new axis will be added... I guess a maximum of 4 axis, 2 right and 2 left should be enough for everyone, what do you think? anyway, if somehow a fifth range is need we go back to rescaling data. (I think it should be a last resource, tough, because you lose the numbers when you rescale data)

griebd, 2015-05-09:
It should be as wanted now!
But before making the pull request I will add a labels to identify with axis correspond to witch graph. The idea is to name the axis L1, L2, R1 and R2 and add those in the legend, like "RPM (L2)", "SWR (R1)". Sounds good enough?

flarssen, 2015-05-09:
I think two legends would be better, one left and one right, and use two different colours to relate axis with graph labels on each side. 
I think two legends would be better, one left and one right, and use two different colours to relate axis to graph labels on each side.

griebd, 2015-05-09:
Will look at two legends, looks like a good suggestion!
But didn't understand about colors... they are already used to identify which graph is which in the legend.

flarssen, 2015-05-09:
I was thinking about text colour of labels and scales. Something like this:
[image]

griebd, 2015-05-09:
now I see, but I think the lib doesn't allow it...
anyway, I think if you use colors for two different associations things start to get messy and difficult to understand!!! imagine you get a third graph on your drawing above with a blue line and a green axis....
having a label for the axis is a lot more obvious and clear, i think... don't you agree?

flarssen, 2015-05-09:
Maybe. As long as there is a maximum of two axis each side and they are sorted by axis in the legends, I think both solutions would work. 
Regarding colours, having graph colours user definable would be nice. Perhaps from the dialog currently used for changing graph labels.

closed 2015-05-22 by griebd

griebd, 2015-05-22:
Should be all good now, just wait for 2.0.18.
User definable graph colors would be a completely new feature. If you really fell the need for it please open a new issue (if open, I will probably try to implement it :wink:).

commit 5a4a8ce4 (2015-05-09): Log Viewer multi-axes. Fixes #2267.
commit 46ccd159 (2015-05-22): Log Viewer multi-axes. Fixes #2267.