opentx/opentx #2268: Companion 2.0.99 crash on write settings to TX

universam1, 2015-05-06:
Bug: When I try to write settings to TX the current Companion 2.0.99 from 5/6/15 crashes.

Tried following:
- Uninstall old companion
- Cleaning - reinstalling
- Clear EEProm settings for TX
- full reset of TX
- updated zadig to 2.1.2

Companion 2.0.15 workes
Write firmware works, reading works, only writing crashes.

Computer Win7 with USB3

TX is Taranis+ with firmware 5/6/15. Same problem was before with oder versions of 2.0.99

Any idea?

tomkfly, 2015-05-06:
2.0.99 is for development use only and is not compatible with 2.0.15.
You should load openTX 2.0.17 to the Taranis and OpenTX Companion 2.0.17 to your computer.

universam1, 2015-05-06:
Sure - I just wanted to mention that 2.0.15 is working okay in comparison to 2.0.99. Therefore its not up to my setup.

bsongis, 2015-05-06:
@universam1 Thanks for your test, I will have a look to this one!

bsongis, 2015-05-07:
RadioData is too big to be instanciated on the stack

commit 2c3eb5e3 (2015-05-07): Fixes #2268
commit a8dba1a8 (2015-05-10): Fixes #2268
closed 2015-05-07 by bsongis