opentx/opentx #2271: DT_REG and DT_LNK macro expansion in Mac build

andyjenkinson, 2015-05-07:
I just did a Mac companion build for 2.0.17. I hit a compiler error due to two macro declarations in radio/src/targets/simu/simpgmspace.cpp
```
In file included from /Users/aj/Sources/opentx/companion/src/firmwares/opentx/opentxSky9xsimulator.cpp:104:
/Users/aj/Sources/opentx/companion/src/../../radio/src/targets/simu/simpgmspace.cpp:710:35: error: expected unqualified-id
        if ((res->d_type == simu::DT_REG) || (res->d_type == simu::DT_LNK)) {
                                  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/sys/dirent.h:128:18: note: expanded from macro 'DT_REG'
#define DT_REG           8
                         ^
In file included from /Users/aj/Sources/opentx/companion/src/firmwares/opentx/opentxSky9xsimulator.cpp:104:
/Users/aj/Sources/opentx/companion/src/../../radio/src/targets/simu/simpgmspace.cpp:710:68: error: expected unqualified-id
        if ((res->d_type == simu::DT_REG) || (res->d_type == simu::DT_LNK)) {
                                                                   ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/sys/dirent.h:129:17: note: expanded from macro 'DT_LNK'
#define DT_LNK          10
                        ^
```
It didn't link the use of simu::DT_REG and simu::DT_LNK, presumably these are expanded already to 8 and 10 by the preprocessor. I changed them to DT_REG and DT_LNK and it compiled, but not sure if this is the correct solution or if it will work on all platforms still.
```
diff --git a/radio/src/targets/simu/simpgmspace.cpp b/radio/src/targets/simu/simpgmspace.cpp
index 1ef61f6..eb56fbe 100644
--- a/radio/src/targets/simu/simpgmspace.cpp
+++ b/radio/src/targets/simu/simpgmspace.cpp
@@ -707,7 +707,7 @@ char *findTrueFileName(const char *path)
       for (;;) {
         struct simu::dirent * res = simu::readdir(dir);
         if (res == 0) break;
-        if ((res->d_type == simu::DT_REG) || (res->d_type == simu::DT_LNK)) {
+        if ((res->d_type == DT_REG) || (res->d_type == DT_LNK)) {
           // TRACE("comparing with: %s", res->d_name);
           if (!strcasecmp(fileName.c_str(), res->d_name)) {
             strcpy(result, dirName.c_str());
```

andyjenkinson, 2015-05-07:
I built it with this modification already:
http://www.open-tx.org/2015/05/04/opentx-2.0.17/

If it is wrong I can rebuild today or tomorrow morning. After that I will be on holiday for 2 weeks.

bsongis, 2015-05-07:
Would you like to push your changes? I will add you as a commiter

projectkk2glider, 2015-05-07:
Careful, with those changes it does not compile on Linux.

andyjenkinson, 2015-06-01:
Can anyone suggest something that will work on all platforms?

andyjenkinson, 2015-07-24:
I don't know enough to suggest a cross-platform solution.

However I just built 2.1.0 successfully, without any link errors, with the simu:: namespace intact. However the simulator is not available in this version, presumably the simpgmspace.cpp object was just never built. Am I right in thinking this was a general issue in 2.1.0?

bsongis, 2015-07-24:
@andyjenkinson would you send me the files you modified? I will make it cross-platform!

andyjenkinson, 2015-07-24:
I haven't modified anything for 2.1.0, it build straight away with no errors. But I suspect it is because the simulator was not built. I found on rcgroups mentioned that this was a bug with 2.1.0, so went looking here - pull #2482 

I will try compiling from master and see what happens. In the meantime, 2.1.0 is up on the website, should I edit the website post?

bsongis, 2015-07-24:
Yes please!

bsongis, 2015-07-24:
@andyjenkinson Would you contact me by email? I would need your help to setup a OSX virtual environment to do automatic builds ... Thanks!

bsongis, 2015-07-24:
Oh now I see your diff, I will apply them now.

commit eb03c6a0 (2015-07-24): Fixes #2271
commit 69cbb064 (2015-07-24): Fixes #2271
closed 2015-07-24 by bsongis