opentx/opentx #2277: firmware won't load into transmeter

Kennyptho, 2015-05-08:
ok, I just downloaded the newest version of Companion OpenTx ... It installed just fine. Then it asked if I wish to load the firmware into my radio... I click ok then a message popped up saying like always  DFU Util executable not found ....

Kennyptho, 2015-05-08:
for what its worth the radio I think was a waste of money

bsongis, 2015-05-08:
Yes if you don't read the manual it will be difficult. Where do you live, here 2 guys are waiting for the Taranis being again in stock, they will be happy to get yours.

closed 2015-05-08 by bsongis

Kennyptho, 2015-05-08:
I not chineese i speak english and read it hell i didn't even get a god damn manual with my radio so blow it out yer ars pall

Kennyptho, 2015-05-08:
ya they can suck ma nuts chuck over 325.oo dallers and they can fight over it

Kennyptho, 2015-05-08:
thats not including the taxs and duties  and deliveries i paid... by the time i got it, it cost me over 400.oo dollars

Kennyptho, 2015-05-08:
fix your fucked up software dickhead

tomkfly, 2015-05-08:
Great way to ask for help

kilrah, 2015-05-08:
The software that your radio comes with is perfectly good and is not supported by us. You have absolutely no reason to change it, especially if you can't read or understand instructions.

ckleanth, 2015-05-08:
Great way to ask for help x2, this is not the radio for you. sell it off, I'm sure someone will be happy to take it off your hands.

griebd, 2015-05-08:
400 dollars? just send me it and I will gladly take it to use as testing platform for OpenTX, so I can help making it even better than it already is!!!!!!
OpenTX and FrSky just changed the RC world for the better!!!!!!!

Kennyptho, 2015-05-08:
you may sell a cheep radio but over all Futaba (
18MZ
18-Channel 2.4GHz Computer System) beets Fr Sky hands down and they have a full illastraited manual simplifed

Kennyptho, 2015-05-08:
SCAM ARTTEST MORE LKE IT

griebd, 2015-05-08:
I do not sell, I'm offering to buy yours for the price you paid! Anyhow, locked too.... just a spammer!