opentx/opentx #2279: pardin my french

Kennyptho, 2015-05-08:
I not chineese i speak english and read it hell i didn't even get a god damn manual with my radio so blow it out yer ars pall
 @Kennyptho
  
Kennyptho commented 9 minutes ago
ya they can suck ma nuts chuck over 325.oo dallers and they can fight over it
 @Kennyptho
  
Kennyptho commented 7 minutes ago
thats not including the taxs and duties and deliveries i paid... by the time i got it, it cost me over 400.oo dollars
 @Kennyptho
  
Kennyptho commented 6 minutes ago
fix your fucked up software dickhead
think i'll go bace to futaba at least they made a real radio and manual that everyone can understand.

closed 2015-05-08 by bsongis

Kennyptho, 2015-05-08:
better hurry up and close the before people find what kind of an asshole you really are

kilrah, 2015-05-08:
You didn't even read the real and correct answer to your question in another one of the issues you opened.