opentx/opentx #2283: unable to set negativ values for logical switch conditions

diablol, 2015-05-09:
It seems unable to set negativ values for logical switch conditions. 
You can do it in companion but not in the transmitter. 

by the way is it normal that 2.0.99 companion at current status is showing very different values in some areas compared to the firmware?

kilrah, 2015-05-09:
Looks like a bug indeed.

> by the way is it normal that 2.0.99 companion at current status is showing very different values in some areas compared to the firmware?

Hmm what exactly? Many things are still broken, of course knowing which will help fix them.

closed 2015-05-09 by kilrah

diablol, 2015-05-09:
> what exactly?
at moment actually everything ;) strange letters and wrong buttons everywhere

[image]
[image]
[image]
[image]
[image]

kilrah, 2015-05-09:
Are you using firmware and Companion builds from the same day? There have been changes recently.

bsongis, 2015-05-09:
I am afraid he doesn't use the nightly builds but the "release" builds which are not up to date at all!

diablol, 2015-05-09:
What do you mean? 2.1 release?
I use the nightlys from here http://jenkins.open-tx.org/nightly-21/
I Install companion and force download over companion and flash it. 

Am I doing something wrong?

diablol, 2015-05-09:
Ok as I recently asked in openrcforum if the download over companion is the same as the one in the nightly directory it seems they are not the same. I just flashed the one from nightly and it appears with the same issues. So I assume they are not the same. Am I right?

kilrah, 2015-05-09:
I am the one who answered you on the forum but only learned right now that no they were not the same, the builds available from companion were not the latest state of the next branch, they needed manual update which hasn't been for a long time. 
That should now be fixed for next builds..

diablol, 2015-05-09:
I know you answered but no problem :) Keep the good work up guys!! Your job is unbelievable. 
As I get used to github I will try to participate more as I do a lot of stuff with opentx.

kilrah, 2015-05-09:
OK it's now fixed, it will be updated nightly and a build has been run so if you now download from companion you'll get the current status (checking for updates will NOT notify you though, you just have to download manually).

Note that due to the same changes you saw in companion you'll have to discard all your radio settings (do a factory reset on the version screen).

diablol, 2015-05-09:
even closed I add here a similar problem

unable to set decimal value e.g. for calculated lowest cell value sensor for logical switches

[image]
[image]

diablol, 2015-05-09:
e.g. for ALT the units for logical switches are always +1 from the telemetrie list
for example meters in telemetrie results in feet for log switches
this all may the same problem

kilrah, 2015-05-09:
Can you provide an eepe file that shows the problem? Would save quite a bit of time recreating one to reproduce.

Might want to open a new issue, it has nothing to do with the original one, best to separate unrelated things.

commit 86f4957c (2015-05-09): Fix #2283
commit 2cde14e9 (2015-05-10): Fix #2283