opentx/opentx #2284: Negative Expo in the mixer + trim => jump from +100 to -100!

fred99, 2015-05-09:
After dealing with expo in the mixer of the Taranis and the Companion under version 2.0.17 I have seen a curious behaviour easily produced in following thoses steps:
1) Set up
[image]
2) The result until 80% it's OK
[image]
3) Above 80% it's NOK
[image]

fred99, 2015-05-09:
I have seen this problem on Taranis 2.0.15 and Companion 2.0.17.

projectkk2glider, 2015-05-09:
Wow this is a serious bug. I will have a look at this now..

projectkk2glider, 2015-05-09:
Confirmed!

If channel source value is >100% or <-100% and negative expo in the same mixer line -> bug as soon as source value exceeds +-100%

bsongis, 2015-05-09:
Tell me if you need help!

projectkk2glider, 2015-05-09:
This has gone unnoticed since 4b9270182e6faa098c5251e6c54360c67a9cc1c2 when parameter type changed from 16bit to 32 bit:
```c
int expo(int x, int k)
int16_t expo(int16_t x, int16_t k)
```

The bug is in the `expo()` when x is less then -1024. In this case `expou()` gets called with enormous number for `x` value.:
```c
int expo(int x, int k)
{
  if (k == 0) return x;
  int y;
  bool neg = (x < 0);

  if (neg) x = -x;
  if (k<0) {
    y = RESXu-expou(RESXu-x, -k);
  }
  else {
    y = expou(x, k);
  }
  return neg? -y : y;
}
```

projectkk2glider, 2015-05-09:
The problem is that we need inverse function of `expou()` if the `k` is negative. 

This is done currently by inverting input value (x) and inverting output value (y). This works as long as input values are in a range from 0 to 1024. If we cross that range, we are in trouble.

@bsongis I have no idea right now how to elegantly solve this. I don't want to limit x to +-1024 before we call `expou()`.

ckleanth, 2015-05-09:
as i understand the issue is only if one uses the expo function. If one uses custom curve no issues  right?
Maybe someone need to inform people in the forums?

kilrah, 2015-05-09:
Only expo, and only *negative* expo which usually has no reasonable use which is why this went unnoticed for more than 1.5 year...

bsongis, 2015-05-09:
And again it's me :(

ckleanth, 2015-05-09:
benard this project wouldn't be where it is now without you lot .

ckleanth, 2015-05-09:
Maybe a short term solution is not to allow -ve value for expo?

bsongis, 2015-05-09:
I will have a look tomorrow. I need a good sleep before :)

bsongis, 2015-05-11:
Limiting to [-1024:+1024] would be consistent with the current behavior in Inputs screen (for example you setup an input with an offset = +50 and an expo = -20

openfsguruh, 2015-05-11:
If I am correct, the problem occurs when input to expo function is smaller than -1024 or bigger than 1024 and expo is negative. Also I am not so sure about a positive expo, are there tries? In the original idea from thus, the expo functions are directly coupled to the input sticks. This made it impossible to have other values than -1024 to 1024, so this expo replacement function was never designed to get free mixer values. 
This seems to be still valid for the old 9x Transmitters as they don't Support the new Input values and don't allow expo in mixer. 
As solution I can imagine following steps:
- A shortterm solution is to Limit Input values from -1024 to 1024, but it brakes the possiblities necessary in mixer.
- A longterm solution is to rewrite expo with the new possiblities the new ARM CPU gives us. Just remember the current expo is in real more or less something like x*x or x*x*x*x very optimized to get an old 8bit CPU the chance to calculate it. The old Transmitter don't need the new possiblities.

openfsguruh, 2015-05-11:
Anyway I have to say I like a lot the new possiblities and always vote for more options.

bsongis, 2015-05-12:
The bug is there since long, no need to rush and find a short term solution. I will vote for a different expo function on ARM boards.

bsongis, 2015-07-17:
The final fix for this one will need some rewrite, which is targetted to 2.2.0

fraternl, 2015-07-17:
I wanted to use negative expo for elevator compensation. Just saw this and am glad I chose to use a 3 point curve.

JapieG, 2016-09-17:
Ouch i use negative expo on an 6 meter glider with slow response to aileron. Is it better to change the programming of the glider?

projectkk2glider, 2016-09-17:
@JapieG Yes, until this issue gets fixed use a curve to get more response around center.

3kouet, 2018-06-08:
I have the same problem, which almost kill me with an helico (joking but could happen). With taranis Qx7S, last opentx firmware. If i put a negative expo in mixer (and nothing in inputs or elsewhere), if trim is up than 2 or 3, funccion jump from +100 to -100...  So i try to not putt anymore curves in mixer, but only in inputs... What i do have difficulty to understand is what happen if you set different curves for a same channel in input, then mixer, then outputs, do they mix or one is prior an other ? Maybe something can be do for avoid thoses multiple mixing ? As a warning or something which avoid multiple setting ?

3djc, 2018-06-08:
What you describe here is different and fixed in 2.2.2 (https://github.com/opentx/opentx/issues/5816)

3kouet, 2018-06-10:
I use 2.2.1, that's the last one i got last week.. I will go right now to companion for got the 2.2.2 ... Thabks a lot for all this work !! You are our gods ;-)

3djc, 2018-06-10:
2.2.2 is not available yet, but 2.2.2RC1 should be released shortly

3kouet, 2018-06-10:
That's what i though !! Not matter, im not using it anymore in mixer, i was getting crazy to do not understand what was the problem with my setting.. I never though it could be a bug. 

So waiting for 2.2.2  !! See you next !