opentx/opentx #2292: Companion loses save for expo setting

Lob0426, 2015-05-11:
I set exponential on my Aileron, Rudder and Elevator controls in companion. Then I save to a document. when I open the models document the expo has reverted to Diff.

When I read the models from my Taranis X9D radio the expo settings were fixed on the radio itself. I read the models and settings from the radio. Then I save this into a document. Stays correct until I close then reopen Companion. Then the settings have reverted to Diff from expo again.

Am I doing something wrong or is this happening for others?
I am using Companion ver2.0.17 on Windows 7 with admin rights

kilrah, 2015-05-11:
Where are you setting expo? inputs or mixer? Can you post screenshots?

Can't reproduce here... are you sure you're opening the correct document?

Lob0426, 2015-05-12:
Sorry about the duplicate.
I am setting Expo in mixes. I just realized I am setting it in the wrong place, it should be in inputs. Maybe the Expo option should not be available in mixes.

From: Andre Bernet 
Sent: Sunday, May 10, 2015 11:06 PM
To: opentx/opentx 
Cc: Lob0426 
Subject: Re: [opentx] Companion loses save for expo setting (#2292)

Where are you setting expo? inputs or mixer? Can you post screenshots?

Can't reproduce here... are you sure you're opening the correct document?

kilrah, 2015-05-12:
The option is available in both for flexibility, there is always more than one way to do something in OpenTX. 
It's not "wrong" to use expo in mixers. 

And from my tests it works, so if it doesn't for you we'd still like more details to see what might need to be fixed...

Lob0426, 2015-05-12:
I think I know the problem. If you do not set a value in the box for Expo it automatically reverts back to Diff. So it probably is user error!

I have set a Expo value in the mixes, saved, closed Companion then reopened Companion and the Document. The setting is now persistent.

I attached a picture.

Living proof you caint cure stupid!

Richard
From: Andre Bernet 
Sent: Tuesday, May 12, 2015 1:49 AM
To: opentx/opentx 
Cc: Lob0426 
Subject: Re: [opentx] Companion loses save for expo setting (#2292)

The option is available in both for flexibility, there is always more than one way to do something in OpenTX. 
It's not "wrong" to use expo in mixers. 

And from my tests it works, so if it doesn't for you we'd still like more details to see what might need to be fixed...

closed 2015-05-12 by kilrah