opentx/opentx #2300: Bug using GVAR in logic switch Companion TX 2.0.17

tomkfly, 2015-05-12:
Using the following settings there seems to be a bug, or am I missing something?
Special Function :SF1/  ON / Adjust GV1/ Source Thr / on (this bit works)
Logic Switch:  L1/ a<b / CH10 / GV1. (Works if Thr is less than 0, but if Thr is more than 0 the Logic Switch drops out).
Mixes: Ch3 [i3] Thr / weight (+100%)
             R /. CH10 / Weight (+100%) / Switch (L1)
             CH10 / SG Weight (+200%) / Offset (-100%) / Slow (u5 : d0)
( Throttle follows CH10 up to middle(0) position and the jumps to the position of the Throttle stick).

tomkfly, 2015-05-12:
Just checked the same settings on V 2.0.14 and the same happens, so maybe I am missing something.

davxlw, 2015-05-13:
Hi,
Don't know if we can call this behavior a bug, but the channels and gvars values are not in the same range in the logical switches, see #1361 for more details.
Bye.

tomkfly, 2015-05-13:
You read a value from a channel and when you try to compare that with another channel they are different ranges? Doesn't make sense.

Tom

kilrah, 2015-05-13:
Not when you compare 2 channels, but when you use GVARs.

kilrah, 2015-12-29:
Duplicate of #1361

closed 2015-12-29 by kilrah