opentx/opentx #2301: File path issue

RuKeBo, 2015-05-12:
I am not sure if this is a big issue but it appears that the "fndTrueFileName function is not working with a "/"forward slash.  The f_open function will open the file fine but it generates and error with the findTrueFileName function.  You can see from the debug output below that it will play the sound file fine, but then it gives an error that findTureFileName can't find it.  Most if not all of these are set in the Taranis.  The first line says that "BOminiH.bmp" is not found but then it opens and closes it in the next two lines, it displays on the simulator as well.  In fact, all this appears to work fine, all the sounds play and all the bitmaps show up.  I have two telemetry scripts that also work but it says it can't find them and gives an error up to "telem5.lua" (I did not paste all that info here).

findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//BMP/BOminiH.bmp)
	not found
f_open(C:\Users\rborn\Documents\Taranis\Clone//BMP/BOminiH.bmp) = 620A3068
f_close(620A3068)
playFile("/SOUNDS/en/SYSTEM/thralert.wav", flags=0, id=0)
findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//SOUNDS/en/SYSTEM/thralert.wav)
	not found
f_open(C:\Users\rborn\Documents\Taranis\Clone//SOUNDS/en/SYSTEM/thralert.wav) = 620A3068
f_close(620A3068)
playFile("/SOUNDS/en/SYSTEM/swalert.wav", flags=0, id=0)
findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//SOUNDS/en/SYSTEM/swalert.wav)
	not found
f_open(C:\Users\rborn\Documents\Taranis\Clone//SOUNDS/en/SYSTEM/swalert.wav) = 620A3068
f_close(620A3068)
playFile("/SOUNDS/en/engdisa.wav", flags=0, id=1)
playFile("/SOUNDS/en/thrdis.wav", flags=0, id=10)
playFile("/SOUNDS/en/batlow.wav", flags=0, id=14)
playFile("/SOUNDS/en/batlow.wav", flags=0, id=15)
eeprom write model
findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//SCRIPTS/BOMiniH/telem1.lua)
	not found
f_stat(C:\Users\rborn\Documents\Taranis\Clone//SCRIPTS/BOMiniH/telem1.lua) = OK
findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//SCRIPTS/BOMiniH/telem1.lua)
	not found

RuKeBo, 2015-05-12:
This also happens in a telemetry script when loading a bitmap to the screen.  In this example, the script uses the backslash notation but it cannot be changed to a forward slash "\" or it will throw an error about an illegal escape sequence.

f_open(C:\Users\rborn\Documents\Taranis\Clone//SCRIPTS/BMP/RSSI08.bmp) = 620A3068
f_close(620A3068)
findTrueFileName(C:\Users\rborn\Documents\Taranis\Clone//SCRIPTS/BMP/battery.bmp)
	not found

kilrah, 2015-05-13:
What is your SD card path set to? Sounds you simply have a stray / at the beginning of it.

RuKeBo, 2015-05-13:
If you mean in Companion and the SD Structure Path, it is shown above ("C:\Users\rborn\Documents\Taranis\Clone").  It appears to me that this is something in the new version, I don't think this happened before.  You can see it is adding an extra "/" at the beginning of the path after the "SD Structure Path" but I think it is adding it to the item path, not SD Card path since it is the same backslash, not forward slash.  Most of the paths for the items like the bitmaps are set in the Taranis, I have no control over them.  But the same thing happens in a lua script (second example).

PR #2342 (2015-05-25): Fixes #2301: findTrueFileName() was not finding any files on Windows
Closes #2301
commit 69b6a987 (2015-05-25): Merge pull request #2342 from opentx/projectkk2glider/issue_2301_findTrueFileName_fix_on_windows_for_next

Fixes #2301: findTrueFileName() was not finding any files on Windows
commit e9c42745 (2015-05-25): Fixes #2301: findTrueFileName() was not finding any files on Windows
PR #2341 (2015-05-25): Fixes #2301: findTrueFileName() was not finding any files on Windows
Closes #2301
closed 2015-05-25 by bsongis