opentx/opentx #2303: SPECIAL FUNCTIONS/Owerr.CH

asykora, 2015-05-13:
Transmitter Turnigy 9XR does not lock the channel.
 (Translated by Google)

kilrah, 2015-05-13:
Please describe the problem better, maybe find someone to help you with English. What radio do you have, what firmware version an with which options, what are you doing and what doesn't work as expected...

asykora, 2015-05-14:
> Komu: opentx/opentx <opentx@noreply.github.com>
> Předmět: Re: [opentx] SPECIAL FUNCTIONS/Owerr.CH (#2303)
>
> "

I have got radio Turningy 9XR-pro and the receiver D8R 2 plus. 

I tried to set the model of plane in the firmware 2.1.0 which will be 
finished soon. 

I cannot set the the locks of channels neither on my radio Tuningy 9XR-pro 
nor by simulation in Companion 2.0.15 and 2.0.17 on the screen of special 
function. 

O want to use this new firmware because the function Alt does not work in 
telemetry in the versions 2.0.15 and 2.0.17 where the function VSPD of the 
mentioned firmwares is good. 

In the firmware 2.1.0 is a new telemetry and in my radio the function Alt as
well as VSPD work well. 

---------- Původní zpráva ----------

(https://github.com/opentx/opentx/issues/2303#issuecomment-101635849). 

"=

asykora, 2015-05-16:
I have got radio Turningy 9XR-pro and the receiver D8R 2 plus. 

I tried to set the model of plane in the firmware 2.1.0 which will be 
finished soon. 

I cannot set the the locks of channels neither on my radio Tuningy 9XR-pro 
nor by simulation in Companion 2.0.15 and 2.0.17 on the screen of special 
function.  

O want to use this new firmware because the function Alt does not work in 
telemetry in the versions 2.0.15 and 2.0.17 where the function VSPD of the 
mentioned firmwares is good. 

In the firmware 2.1.0 is a new telemetry and in my radio the function Alt as
well as VSPD work well

---------- Původní zpráva ----------

(https://github.com/opentx/opentx/issues/2303#issuecomment-101635849). 

"=

kilrah, 2015-05-16:
2.1 is absolutely not safe to use at this point.

Many things are still broken especially on platforms other than Taranis, and your model setups can break overnight.

Alt should work in 2.0, if not please open a separate issue with all details about what sensors you're using.

Leaving this issue open as a reminder that navigation in SF is broken on "small screen" platforms.

closed 2015-06-28 by bsongis