opentx/opentx #2304: Does not recognize sd card when pluged in to computer

rako1, 2015-05-14:
Hi I just Receieved my Tranis plus and flashed it to 2.0.17 when I finished I went to eject the mass storage and noticed the frsky tranis bootloader was not highliter but the removable disk and the Tranis disk were. So i ejected both of them to close them out to unplug my radio. Now I nitice when I plug my radio back in to access the SD card only the bootloader came up nothing else can anyone help me

tomkfly, 2015-05-14:
Are you switching on the Taranis, using the three finger method, ( both horizontal trims held towards the power switch as you switch on ) before you connect the USB?

rako1, 2015-05-14:
Yes

rako1, 2015-05-14:
It only shows bootloader

kilrah, 2015-05-14:
Try to remove/reinsert the SD card, and check it's available on the radio itself.

rako1, 2015-05-15:
ok when I turn on my radio in bootloader mode 3 finger trim switch then plug in my usb cable both disk pop up tranis and removable is this right? and thats how I access my sd card?

kilrah, 2015-05-15:
Yes... One is the SD card and one is the firwmare and models/settings

closed 2015-05-15 by kilrah