opentx/opentx #2305: Wrong unit for airspeed S.Port telemetry data

pawelsky, 2015-05-14:
According to the S.port protocol spec (v3.1) the unit for airspeed (appID 0x0A00~0x0A0F) is (km/10)/h, Taranis radio firmware (2.0.17) however expect it to be knots/10 which causes invalid airspeed values to be displayed on the telemetry screens when the data is passed as required by v3.1 of the S.Port specification.

I believe this is done to support the unofficial N2 AirSpeed sensor, but since FrSky is now releasing the official ones (ASS-70, ASS-100) maybe it is worth looking deeper into this issue.

I've asked Frsky via their DevUniom forums what is the unit for the data sent by the official airpseed sensor ((km/10)/h as in spec or knots/10 as in Taranis firmware) but did not get a reply yet.

kilrah, 2015-05-14:
By the time the frsky sensor actually becomes available opentx 2.1 will likely be out, and that has configurable units.

pawelsky, 2015-05-14:
Probab;y yes, but do I read it right that in 2.1 the default unit for airspeed is defined as m/s (not km/h)?

>  { AIR_SPEED_FIRST_ID, AIR_SPEED_LAST_ID, ZSTR_ASPD, **UNIT_METERS_PER_SECOND**, 1 },

kilrah, 2015-05-15:
Hmm indeed, that dates back from before FrSky set an unit for that themselves.

Changing that to match FrSky's decision would break the N2 sensor (or require a firmware upgrade for it)
Alternatively someone could talk to FrSky and have them change their unit to match the N2 sensor.

Bruneaux, 2015-05-27:
I believe that the N2 air speed sensor will not require a firmware update if there were a way to enter and absolute number into the "New Sensor" - "Calculated" - "Multiply".  I've looked and I can't find a way to do a basic "multiply  ASpd  x.xx"  or even by a gvar.  That would be very helpful.  I wanted to do some basic math (add or multipy) on other sensors but it appears that all math must be done with two sensors (ie.  multiply current x voltage).

Bruneaux, 2015-05-28:
To be clear,  the N2 air speed will not require any firmware updates because the OpenTX team that is working on the new telemetry has implemented a 'ratio' function.   It will be easy to tell a user what ratio is necessary to get from the native (whether it is knots, kph or M/s) to the desired units.

  I think that the issue could be closed.

kilrah, 2015-05-28:
It's still a "dirty fix" though. 

One of the following solutions would be the clean way to do it:
- Change FW on the N2 sensor to match FrSky's choice
- Change FrSky's choice to match the N2 sensor's unit (probably not possible, I imagine FrSky have already coded their other accessories e.g. smart dashboard according to their own unit)
- Change the data ID on either the N2 unit or the FrSky one, so both are recognised as different sensor types with their matching unit. Might have the same problem as no2 above

Bruneaux, 2015-05-28:
It just looks like there is no real 'answer'.  There has to be some give on one side or the other.  Recalling all of the sensors that have been sold over 14 months is not an answer.  They are all over the world.
  FrSky has probably already shipped their version to the whole world, too (don't see any of them in the US yet).  And they are not likely to yield to an after market device.
  Mitigating the problem ( for N2 ) by directions that will correct the value from Knots to KPH is probably the best we can ask for.  FrSky owns the market.  Further, having N2 modify any further firmware to incorporate KPH  would be recommend if KPH is, indeed, the new units that are being coded.  The start of the thread indicated M/s, which would require both to do conversions.

Bruneaux, 2015-05-28:
Oh, and lets not forget that OpenXSensor is coded to Knots.  Those that have made their own sensors will have either re-flash or use the conversion ability.

pawelsky, 2015-05-28:
The official unit from the spec is km/h (or rather (km/10)/h to be precise), not m/s.

I believe aligning to whatever Frsky have chosen would be the way to go (it is their protocol and spec after all). For backward compatibility there could be a FW variant with N2 support or ratio function could be used.

kilrah, 2015-05-28:
> I believe aligning to whatever Frsky have chosen would be the way to go (it is their protocol and spec after all)

Agreed.

>FrSky has probably already shipped their version to the whole world, too (don't see any of them in the US yet). And they are not likely to yield to an after market device.

Haven't seen any anywhere yet, and most importantly FrSky equipment can be reflashed by the user.

The N2 sensor should also be user-reflashable, I don't know if it is but if it isn't it's a serious design overlook. With today's dynamics anything that runs firmware and interoperates with other stuff should be user-reflashable, especially things that piggyback on 3rd party system as things can break overnight.

kilrah, 2015-07-07:
It turns out that the N2 sensor can be reflashed, so a unit change would be doable. 
I.e. we should change the unit to match the FrSky protocol.

Bruneaux, 2015-07-07:
> No virus found in this message.
> Checked by AVG - www.avg.com <http://www.avg.com>
> Version: 2015.0.6037 / Virus Database: 4365/10181 - Release Date: 07/07/15

Andre,

   Yes, I had that conversation with Kilrah weeks ago.

   There really is not need to re-flash the AirSpeed sensors because it 
is easier to have the use change the "Ratio" to match the new value.

   I am prepared for a flood of questions,  and changing the "Ratio" is 
the easiest way to deal with the issue.

   FrSky has held back the release of their air speed sensors.  Maybe 
they are re-thinking.

   Either way, with a "Ratio"  (given it has enough precision) either 
air speed sensors can be dealt with.
   Correct?

Chuck

kilrah, 2015-07-09:
The problem is that as this issue got somehow stuck, and as such the ratio currently only has one decimal. That wouldn't be suitable, and as we're about to release we can't really change it anymore.

Bruneaux, 2015-07-09:
1 Knot = 1.8519984 Kilometers Per Hour

  So I need to get a "ratio" that will be as close as possible.

  Can someone send me the equation for "Ratio"?  I don't know if it linear or log or some other equation.  I know that Andrea was working on that code.

  I will deal with my customers as they discover the change.  

Chuck

projectkk2glider, 2015-07-09:
I see no problem. Here is my solution:

* configure N2 speed in knots
* create a new sensor, type CALCULATED, Multiply, unit kph, source1: N2 speed

The second sensor will convert from knots to kph.

I've just tried it Companion simulator and it works, A1 was used in this example as speed sensor:
[image]

projectkk2glider, 2015-07-09:
Another possibility is to set the **ratio** for N2 sensor. Here is example with 10.3 knots received from telemetry:
[image]

The ratio formula is:
```c
    value = (custom.ratio * value + 122) / 255;
```

Bruneaux, 2015-07-09:
Thank you for taking time to think about the issue and posting the information.
  The "Calculated" sensor was my second choice and after I spoke with Andrea he indicated that going straight  to the  "Ratio" would be easiest.  I have to agree.
   The ratio seems rather high, but it the math is correct then the math is correct.
    I just don't see this as being a horrible thing.  Easy instructions for someone to follow.

Quote:
 Bruneaux commented on May 28 
Mitigating the problem ( for N2 ) by directions that will correct the value from Knots to KPH is probably the best we can ask for.

pawelsky, 2015-07-25:
Coming back to my question asked earlier in the following comment
https://github.com/opentx/opentx/issues/2305#issuecomment-102163736

Is there are particular reason to have the default unit for airspeed set to m/s? It neither matches the Frsky spec (which uses km/h/10) nor the N2 sensor which uses knots.

I believe it should be set to whatever FrSky chooses.

kilrah, 2015-08-01:
km/h it is.
The ratio given by N2 on their page will need to be updated.

BTW:
> I see no problem. Here is my solution:
configure N2 speed in knots
create a new sensor, type CALCULATED, Multiply, unit kph, source1: N2 speed
The second sensor will convert from knots to kph.

That doesn't work (well it works but gives a wrong value), because

> configure N2 speed in knots

As the received value is expected in m/s this causes the value that is already received as knots to undergo an undesired m/s -> kt conversion.

closed 2015-08-01 by kilrah

Bruneaux, 2015-08-01:
Kilrah,

Thanks for the update.

Chuck

Bruneaux, 2015-08-01:
Hmmm,  I have missed something.

  Why is there any conversation about converting Knots to Meters/second?

Chuck

kilrah, 2015-08-01:
Forget that, it only applied to the solution @projectkk2glider proposed, I'm explaining why it wouldn't work. If the selected displayed unit is different from the hard-coded "expected" one then a conversion is done.

athertop, 2016-12-07:
It appears that this never got resolved, as the current src code for OpenTx now has the default (expected input units) for airspeed to be knots (kts). So it moved from m/s to kts at some point since Pawlsky's original example code posted above. So it still does not meet FrSky spec of kph. To be honest I think m/s would have been better left than changing this to the current kts unit.

kilrah, 2016-12-07:
Someone made tests (documented in another issue) with different data sources and determined that regardless of what the docs say the current code is what gave correct values.

commit 44c096c3 (2015-08-01): Align airspeed units to S.Port spec (Fix #2305)