opentx/opentx #2306: PPM internal module not reachable through GUI

cmarzer, 2015-05-14:
There is a TARANIS_INTERNAL_PPM build option that seems to work but this option is not reachable yet through the GUI

kilrah, 2015-05-14:
What would be your use for it?

bsongis, 2015-05-14:
I want to remove this option, the timer used there is already used somewhere else if I am not mistaken

bsongis, 2015-05-14:
Comment removed, I just checked, and there is no issue on first glance

cmarzer, 2015-05-14:
I also looked quickly through all the code and I didn't see any other use of that specific timer's channel. I know things can be sometimes tricky with STM32 though. 
kilrah : I replaced the internal frsky module and replaced it with a jeti module. It makes a very attractive second transmitter if you are equipped with jeti stuff and don't want to invest in a pricy second DS-16 ! The next step is to update the jeti telemetry stuff (not so easy as it seems with the STM32...but I have a few ideas how to do that part...)

bsongis, 2015-05-14:
You have some comments to fix in the code review, otherwise it's all good

kilrah, 2015-05-15:
> I replaced the internal frsky module and replaced it with a jeti module.

OK. Could have been put in the external module slot too though ;)

> I also looked quickly through all the code and I didn't see any other use of that specific timer's channel. I know things can be sometimes tricky with STM32 though. 

Also check auxiliary things like the DMA channel that is used, and timer/DMA use across all platforms. I remember we had some reserve recently and nearly removed the whole TARANIS_INTERNAL_PPM code due to potential conflicts but I don't remember what now... Need to be extra careful.

bsongis, 2015-06-12:
This one is done, right? Otherwise feel free to reopen

closed 2015-06-12 by bsongis