opentx/opentx #2309: [Bug][OpenTX 2.0.17] Curve not linked with setting in INPUTS menu

LapinFou, 2015-05-15:
Hi folks,

I don't know if it is a bug or not, but it was quite confusing for me.
Most of my setting are done through Companion, so I never noticed before.
I was on the field doing some EXPO adjustment on my Yak55.
I got a triple rate, so it means for each input, I got 3 lines with a different ratio and expo.
[image]

Here is what I've done:
1- Go the INPUTS setting.
2- Select and edit the "MOYEN" rate (2nd line)
3- Change the expo parameter.
Then I realized the curve on the right was not changing.
[image]

In fact the curve doesn't reflect the current settings which are modified, but show the current cruve applied on the input.

So I toggled from SD^ to SD- and I could at least visualize the modification I was doing.

kilrah, 2015-05-15:
The graph always shows what's currently active, it's a feature that's always been there and is also done that way on several other radios. That way you can toggle your switch and see the difference "live" on the graph as it changes.

closed 2015-05-15 by kilrah

LapinFou, 2015-05-15:
Thanks. Sorry for that. I wasn't aware.
:)