opentx/opentx #2311: Add ST7920 LCD support for MEGA2560

olterion, 2015-05-15:
I have written a driver for the ST7920 LCD controller. This driver takes the data structure of the ST7565R driver and merges it into the data structure for a ST7920 driver.
I think it should be possble to get this code into the source code and Makefile. Then everyone can choose the LCD for DIY radios. This type of LCD is very cheap and easy to get.
I have all necessary informations posted in the openrcforums:

http://openrcforums.com/forum/viewtopic.php?f=45&t=6919

I would be very happy, if someone can make my code chooseable for the MEGA2560 DIY radio.

Best regards
olterion

bsongis, 2015-05-15:
@JivaroFAD Franck I think you will agree ;)

JivaroFAD, 2015-05-15:
Yes, that's a good idea ;-)
I've already take a look at this driver early this week, and I will merge it in /stock/lcd_driver.cpp + makefile next week.

@olterion :
- to facilitate code portability, "OUT_C_LCD_RS" will be rename to the existing "OUT_C_LCD_A0" that has the same function (register select)
- "for (xcoord_t x=0; x < 16; x++)" seems to be wrong, is it "or (coord_t x=0; x < 16; x++)" ?
- in lcdInit() :  can I remove the delay after "lcdSendCtl(0x30);" ? There no delay on ST7565P and R dirver in this sequence.

bsongis, 2015-05-16:
Don't forget to commit in a branch so that we can have a code review!

olterion, 2015-05-16:
Hello JivaroFAD,

maybe it's possible to write a comment to the renamed pin (RS->A0), so everyone can see how to connect a ST7920 LCD.

the variable type xcoord_t was for the master branch of opentx. In my current version of the ST7920 driver is had renamed to uint8_t. So you can change xcoord_t to coord_t or uint8_t.

The delays are out of the datasheet for the ST7920 controller.
Some delays are a little bit too long, because of testing.
I have tested the following code is okay:

inline void lcdInit()
{
	LCD_LOCK();
	PORTC_LCD_CTRL &= ~(1<<OUT_C_LCD_RES);  //LCD Reset
	_delay_ms(1); 		
	PORTC_LCD_CTRL |= (1<<OUT_C_LCD_RES);
	_delay_ms(40); 	
	lcdSendCtl(0x30);	// set 8-bit interface
	_delay_us(100);		
	lcdSendCtl(0x36);	// Repeat with graphics bit set to on.
	_delay_us(80);		
	lcdSendCtl(0x0C);	// display ON, cursor and blink OFF
	_delay_us(80);		
	lcdSendCtl(0x01);	// clear display, reset address
	_delay_us(80);
	lcdSendCtl(0x06);	// display ON, no cursor
	_delay_us(80);	
	LCD_UNLOCK();
}

After a reset we have to wait still long and after every control comman we have to wat 72µs. I have choosen 80µs because a little safety.

Thank you for supporting my code and my LCD.

Best regards,
olterion

JivaroFAD, 2015-05-17:
Great ! That will simplify the lcdInitSequence loop in the new lcd_driver :)
For the pin label, please take a look in the updated schematic : http://jenkins.open-tx.org/MEGA2560/MEGA2560_sch.pdf

olterion, 2015-05-17:
Hello,

the pin C7 (LCD_RD) is on the ST7920 for the serial clock line (in 8bit parralel mode)
pin C6 (LCD_WR) is for read and write command (0=write, 1=read)
the pin C3 (LCD_CS) is unused (at least for my LCD)

maybe you can add this informations to the schematic

I have this LCD:
http://www.ebay.de/itm/1-Stuck-5V-12864-LCD-Display-Blau-Modul-Hintergrund-128x64-Punkt-Grafik-Matrix-/351211463745?pt=LH_DefaultDomain_77&hash=item51c5d5a441

maybe you can add this informations to the schematic

JivaroFAD, 2015-05-17:
Right, I will update schematic :)
I've make a first integration of your driver in 744c797 , branch jivarofad/M2560, please could you verify ?

JivaroFAD, 2015-05-17:
Documentation updated : a838344

olterion, 2015-05-18:
Hello JivaroFAD,

I tested your branch on my board and it works!
Additionally I tested this:

 #ifdef LCD_ST7920
    delay_2us(); 
  #else
    delay_2us();       //TODO : verify if can be set to 1ms => so some value for both screen and no #ifdef
  #endif

It works too. The 1ms was only in my mind to wait a litte after the reset line pulling down. This was not in the datasheet of the controller.

It is possible to get my name into the Authors list?
first name: Oliver
last name: Stefan
This is my first code, which I get in the web, so I would be very proud to be part of this software :-)

Thank you!

Best regards,
olterion

JivaroFAD, 2015-05-18:
Perfect, thanks :)
Bertrand will tell you for the authors list.

bsongis, 2015-05-24:
I will check the authors list later, many names are missing

bsongis, 2015-05-24:
Done

closed 2015-05-24 by bsongis

alberiolima, 2018-10-17:
[image]
Thanks!!! atmega128 with lcd st7920