opentx/opentx #2318: Subtrim = trims in Companion

tomkfly, 2015-05-17:
Not urgent, but would be useful if it was possible to incorporate Subtrim = Trims in the Servos page.
It works temporarily in the Taranis Simulator, but is not retained when the Sim is closed.

kilrah, 2015-05-18:
Frankly that should only be done on the radio with the model powered so you see "live" that there is nothing wrong happening on your control surfaces.

tomkfly, 2015-05-18:
I understand it could cause problems if incorrectly used, so probably not a good thing. I will close the request.  The reason I asked was when working on other peoples EEPEs they often have trims well out and it is confusing checking mods using the servo monitor.  I can work around it, so not really a problem.
Personally I always make adjustments mechanically and never use the facility in real life.

closed 2015-05-18 by tomkfly

griebd, 2015-05-18:
I normally don't use others eepes, so I may saying something completely wrong!!
But maybe we could add a helper in companion to remove all trims for cases when you get others eepes... like a button in the model edit first screen "remove all trims".

kilrah, 2015-05-18:
Again that function is available on the radio, which is where it makes most sense...

tomkfly, 2015-05-18:
I asked because I get a number of requests, by PM and Email, to look at and set functions for others. These often have large amounts of trim, which can make it clumsy checking modifications I make. It is not a problem, I can centre the trims using the double click method and copy any changes, less trim changes, to a copy of the original model settings.
I don't have their transmitters but I do suggest they modify the mechanics, if they can, and also explain how to use the trim to subtrim function.