opentx/opentx #2319: LCD AVR: strange y-coordinate dependent bounding line

xmxwx, 2015-05-18:
I couldn't find any explanation for this line of code: https://github.com/opentx/opentx/blob/next/radio/src/gui/9X/lcd.cpp#L1255

I wrote a simple test:

    TEST(Lcd, InversSteps)
    {
      lcd_clear();
      int x = 0, y;
      for(y = 0; y < 16; y++) {
        lcd_putcAtt(x, y, 'B', INVERS);
        x += FW + 2;
      }

      EXPECT_TRUE(checkScreenshot("invers_steps"));
    }

...and it yields strange result on AVR:
[image]

I've traced this piece of code with git blame and here's how the test result looks like before it was edited in ef28d7e:
[image]

CPUARM variants (9XRPRO and TARANIS) use the new code which render without such a strange effect:
[image]
[image]

The mentioned line of code looks like an unused hack. Do you think it's safe to remove it?

bsongis, 2015-05-19:
What do the gtests say?

mhotar, 2015-05-19:
This has been written this way on purpose, to save some flash on the AVR platform.
At first we had no top black line on all inverted characters, I added it. But for AVR it is there only for those characters that match the LCD bytes, it means top inverted pixel fits to the same byte, so we avoid to calculate address/test for LCD boundary, read +1 byte, mask and write back. For AVR we do not need to write text to the lines that cross byte boundary.

mhotar, 2015-05-19:
It seems it can be fixed (if needed) at cost of 44B by replacing this problematic line by
```
        if (inv && y) {
          if (ym8) {
            mask = (0x01 << (ym8-1));
            *p |= mask;
          }
          else {
            uint8_t *r = p - LCD_W;
            *r |= 0x80;
          }
        }  
```
As I am not good in programming at all, you can probably find better solution :)

xmxwx, 2015-05-19:
> Yes it seems wrong to me as well. What do the gtests say?

There are no gtests that cover y > 1. There are only tests for y = 0 and y = 1 (Invers_0_0 and Invers_0_1 respectively) and these cases are handled correctly by code both before and after ef28d7e2333e4b26febcf1c0e3497c765de357eb. I'll replace these tests with InversSteps, which covers 0 <= y < 16.

xmxwx, 2015-05-19:
@mhotar, thanks for explanation and the fix. My fix (not yet pushed) is almost exactly the same. However, I didn't pay so much attention to FLASH usage. I'll see if I can squeeze it even more.

xmxwx, 2015-05-19:
@mhotar, you've inspired me to think about FLASH compactness :-). I've implemented a different approach to 5x7 font rendering and thus achieved the fix which adds only 12 bytes, see #2322. Comments are welcome!

closed 2019-08-19 by kilrah