opentx/opentx #2321: Change Pull-Up resistors for MEGA2560 RF_Activated and DSC_Activated

olterion, 2015-05-19:
Hello,

I had build up the power-circuit of the MEGA2560 schematic PDF.
There are two 100kOhm restistors to pull down the RF_Activated and DSC_Activated signals.
In the board_mega2560.cpp the internal pull-up resistors of the MEGA2560 are activated for this input pins.
In the datasheet of the MEGA2560 on page 355 is the descition of the internal pull-up resistors. Their value is from 20kOhm to 50kOhm. 
So if there is no connection in the jack and the switch is off, we have on both pins a voltage divider with 20...50kOhm to 100kOhm. So we have a voltage on this inputs of 3,3V...4,1V.
This will not be recognized as a 0 in the MEGA2560 and it will not shut down.

I changed the board_mega2560.cpp
So I changed the board_mega2560.cpp:
Line 51: DDRH = 0b00011000; PORTH = 0b10010110; // 7:N/A, 6:RFPw, 5:JackPres, 4:HoldPw, 3:Speaker, 2:N/A, 1:N/A, 0:Haptic

and

Line 101: PORTH &= ~0x10; // PortH-4 set to 0

With this changes my power circuit works properly.
Maybe I'm wrong, but if not I would be glad to get this change into the code.

Best regards,
olterion

JivaroFAD, 2015-05-19:
Hi,

I would prefer to keep pull-up on to have DSC_activated and RF_Activated at ON state by default (without power-circuit).
A simple way to solve the issue is to use 6.8 k resistor instead of 100 k :
- if batt connected (6 - 8.4 V range) : pin voltage = 3.4 - 4.8 V  => pin state = 1
- i batt not connected : pin voltage = 0.6 - 1.2 V  => pin state = 0

Regards,
Franck

olterion, 2015-05-20:
Hello JivaroFAD,

I expect, your suggestion will not work.
In the datasheet of the MEGA2560 on page 355 is the input voltage of a 0 and 1 described:
Input Low Voltage: min: -0,5V max: 0,3V
Input High Voltage: min: 0,6*Vcc = 0,6*5V=3V max: Vcc+0,5V=5V+0,5V=5,5V
So in the area of 0,3V to 3V we are in a not specified range.

That's only possible to choose a pull-down resistor with about 1kOhm.
Then we are sure in the specified range below 0,3V.
But then we have an transverse current of about 5mA, which is too high for only get this switch information. (my opinion)

Maybe there is another solution for this issue?

If anyone will build this DIY TX we should make a note in the schematic to connect this two inputs to 5V for testing. In the normal use you should have the power circuit anyway, because otherwise the radio does not start up correctly.

PPM IN/OUT switch:
Do you mean the circuit with 4066 IC?
That I don't build at the moment. I will not use this part of the radio in nearest future.

Best regards,
olterion

JivaroFAD, 2015-05-20:
Datasheet indicates that max input low voltage 0.3Vcc (not 0.3V), so 1.5V with Vcc = 5V.
A 5.1k resistor can also be used (high voltage : 3V to 4.2V, low voltage : 0.5V to 1V) for testing.

Yes, I mean 4066 IC. I made a mistake on the prototype (reverse voltage :( ) and so did not yet test this feature, which was copied from 9x stock board.

olterion, 2015-05-20:
Hello,

sorry for my mistake!
I didn't recognize, that 0,3 is a factor of Vcc.
I will try this solution.
But then the schematic has to be updated with the correct resistors.

Best regards,
olterion

JivaroFAD, 2015-05-20:
No problem :)
Schematic has been updated yesterday in 28812997ab

I wait from your testing before closing the issue.

olterion, 2015-05-20:
Hello,

I've tried the solution with a 5,1kOhm pull-down resistor. It works fine.

Best regards,
olterion

JivaroFAD, 2015-05-20:
Many thanks, that's great :) :)

closed 2015-05-20 by JivaroFAD