opentx/opentx #2326: Google Earth export

flarssen, 2015-05-21:
Lat/lon data tends to be out of sync. The export from logs seems to pick samples when there is a change in lat/lon, even if just one of them has changed. However, when the next coordinate get updated in the next sample, this seems to be ignored. 
This leads to stepped tracks. Some filtering would be nice. Like; if just one coordinate has changed, pick the next sample (which seems to contain both, even at 10Hz sampling).

Log file:
http://www.rcgroups.com/forums/showatt.php?attachmentid=7887955&d=1432153024

[image]

projectkk2glider, 2015-05-21:
I don't think the problem is in Companion. Your log file already has data in the wrong order. 

Let me explain how this happens:
* GPS data gets received by the radio, each packet contains different part of GPS position. The packets are sent with some time between them. 
* your logging interval is very short 100ms. 
* in your case there are many times where a new line is written into log file, but the GPS data is composed of two GPS locations (old and new). So for example latitude is still from previous GPS position report and longitude is already from the new GPS position.

You will find that if you use bigger logging interval on the radio, you will have less problems of this kind.

I think this can be fixed on the radio. We would have to receive GPS data into a separate structure and only update `frskyData.hub.gps*` data when we are certain all data from new GPS position was received. @bsongis @kilrah is order of GPS data for FrSky GPS sensor defined/fixed? Do we know which item is the last one? On the other hand the GPS sensor itself could already be sending latitude from one measurement and longitude from another. In that case we can't do anything.

projectkk2glider, 2015-05-21:
@flarssen how did you got this log file? It looks like a result of simulation.

kilrah, 2015-05-21:
> @bsongis @kilrah is order of GPS data for FrSky GPS sensor defined/fixed? Do we know which item is the last one?

Not that I know, AFAIK we don't have a way of knowing if something we receive is "the one that matches the previous one of the set" or "the first one of the next set"... i.e. we can't do anything about it.

flarssen, 2015-05-22:
The log is from a "load test" done with Arduino putting out current, voltage, GPS (lat/lon/alt/speed/course/date/time), rpm, temp x 2, vario (alt/vspeed) and 12S cell voltages.
But I see the same with my Arduino based GPS as the only SPort source, and others have reported the same behavior on RCG.
If there is no ID grouping the data, I think it's better to log them as fast as possible and filtering during the export.
@kilrah:
My log shows that only one out of 10 samples/s. is affected, and that update rate for the GPS data is close to 2Hz, even with the high amount of data sent, so 0.5s sounds too long. .

kilrah, 2015-05-22:
>  and that update rate for the GPS data is close to 2Hz, even with the high amount of data sent, so 0.5s sounds too long.

Update rate of the GPS is whatever your GPS decides to send updates at... I was basing myself on the FrSky GPS that sends one lat/lon update per second and sends them "grouped" in time. 

As you're not using the "standard" gear basically everything is dependent on how you do stuff with your arduino.

flarssen, 2015-05-26:
How is telemetry data received from the RF module? I see the same problem when using hub protocol with a D4RII. I send all GPS related data as Frame2, terminated with 0x5E. I would assume this frame is kept intact along the whole transmission line? Of course one would have to wait for the start of the next frame to realize that the previous one had ended, which would add to the overall delay. Really don't understand why FrSky didn't add an "end of frame" ID to the protocol (0x5E followed by ID).

kilrah, 2015-05-26:
As I mentioned on the forum in answer to your question the RF module sends telemetry data to the radio through its smart port. When using D equipment it converts the old stream/data format to smart port format, so the "D" transmission line ends in the module, which uses closed firmware so we have no idea how the conversion is done and what it expects.
That's why I told you you need to do your own testing to see how much you can "deviate" from the documented protocol. Maybe it absolutely wants frames as per the spec with the documented contents, maybe not...

flarssen, 2015-05-27:
Sorry, I forgot that part. I see no difference in Taranis, when sending hub frames or single values.

So back to my initial suggestion for a change in google map export: Skip a log file entry if only latitude changes, and the next two entries contains the same new latitude with a new longitude, or vice versa.
This should give smooth tracks without sacrificing log resolution for data that updates faster

[image]
.

kilrah, 2015-05-27:
That would get rid of the "steps", but may give you a constant error in return.

mox17, 2015-05-27:
Isn't the cause of the problem that latitude and longitude telemetry
packets are sent separately from the receiver, thus causing two log entries
for each real GPS update? (this is at least true for S-port GPS telemetry).
So if only every 2nd entry is logged, or limited to logging on latitude OR
longitude updates, that would take care of the problem.
It should still be possible to get a correct log when flying due
north-south or east-west, so do not suppress more than one identical
latitude or longitude entry.

kilrah, 2015-05-27:
Not necessarily 2, an unknown number that depends on both whatever the GPS wants to send and the user's logging rate.

If you log only the updates there's no difference form now.

If you log only when both change, you may get a shift as you would be "assuming" a of values goes together when they may not.

flarssen, 2015-06-01:
I have a similar problem when using the tinyGps library. I sent the data to the Rx whenever tinyGps.encode() returned true. Later I discovered that it did so for each Gps sentence that were completed, so I was actually sending data twice for each GPS fix. Now I check the interval between each positive return from tinyGps.encode(). If above a certain threshold, I discard  the first one, and consider the second one to be the first of the two sentences it processes. That way, hopefully, all data are from the same Gps fix.

kilrah, 2015-06-01:
Good. The radio will still receive them separately though and not be able to know what goes with what.

projectkk2glider, 2015-06-16:
With the fix in #2366 the @flarssen track has now only 37 points (from 200 log points) and looks like this:
[image]

flarssen, 2015-06-16:
Thanks, quite an improvement!
I have another logfile from real world testing. The other strange part with this one is that it tracked fine the first time i did it (first picture), but later it would drop out in periods, like the end (second picture). No changes to software. Perhaps it works with this fix, or maybe it's a google earth issue?

http://www.rcgroups.com/forums/showatt.php?attachmentid=7965700&d=1434300667

[image]
[image]

projectkk2glider, 2015-06-17:
@flarssen Your new track with my improvements looks like this:
[image]

The missing part you see in  your example is due to the error in GPS altitude. The part of the track that is missing is actually below the terain level. You have to adjust the track elevation properties in GE from `absolute` to `clamped to ground`.

flarssen, 2015-06-17:
Right, I remember doing that, but obviously just the first time.
Thanks again for your great work.

PR #2366 (2015-06-17): Fixes #2326: Google Earth export GPS glitch filtering and lat/lon syn…
…chronization

Closes #2326 

Done a couple of related things:
* a little code refactoring
* fixes #2326 
* filters out any GPS glitches and zero lat/lon location
* fixed icon path
commit 31a651ad (2015-06-17): Fixes #2326: Google Earth export GPS glitch filtering and lat/lon synchronization
commit 7522f976 (2015-06-17): Merge pull request #2366 from opentx/projectkk2glider/issue_2326_ge_filtering

Fixes #2326: Google Earth export GPS glitch filtering and lat/lon syn…
commit c0afcb51 (2015-06-19): Fixes #2326: Google Earth export GPS glitch filtering and lat/lon synchronization (ported from master)
closed 2015-06-19 by projectkk2glider