opentx/opentx #2337: [9X / MEGA2560] OpenTX doesn't start if HELI option is not selected

bsongis, 2015-05-25:
http://openrcforums.com/forum/viewtopic.php?f=45&t=7052&p=97568#p97568

JivaroFAD, 2015-05-25:
Some problem detected on my boards, with 2.0.99, but only with FrSky telemetry connected to the board. Another board without telemetry seems to work well whatever PPM or Heli options.

JivaroFAD, 2015-06-14:
Make exhaustive tests today on M2560 board and stock LCD (KO is only at flash, with always the some avrdure error : 0x07 != 0xff) with 2.0.99 :
[image]

Very strange...

kilrah, 2015-06-29:
> we have just a verification mismatch after flashing (with always the same avrdure error : 0x07 != 0xff)

At what address? There have been known problems with the firmware of some AVR programmers that would mess up when programming beyond the first 128K if I remember well.

closed 2019-08-19 by kilrah