opentx/opentx #2338: Companion QCheckBox backupEEprom can not be disabled

JivaroFAD, 2015-05-25:
All in the tittle !

This is problematic with Arduino M2560, as initial EEprom is not valid (by default in the wring size : 2048 bytes againts 4096).

kilrah, 2015-05-25:
The default state is enabled because of people not thinking of making backups until it's too late. Shouldn't be a problem for M2560, the first backup would just be unuseable.

What's exactly the problem now? If it can be fixed by people selecting a backup folder it seems the issue is rather that there should be a valid default in there.

JivaroFAD, 2015-05-25:
The problem is that the checkbox is shown as not checked, although the default state of the class is enabled.

JivaroFAD, 2015-05-25:
If i'm not wrong, in flashfirmwaredialog.ui, there's an inconsistency between :
widget class="QCheckBox" name="backupEEprom" : property name="enabled", bool = "true"
and 
Backup and restore Models and Settings : property name="tristate", bool = "false"
We should have "true" and not "false" to have the checkbox checked by default ?

JivaroFAD, 2015-05-26:
New tests today, after this mod : 1dc10122ea637a272f0d9bd0b365a49a792e4937
Not very concluant : checkbox "Backup and restore Models and Settings" in Flash Firmaware panel is not checked by default.

An another strange thing : this checkbox is not linked to the checkbox "Enable automatic backup before flashing firmware" in Settings/Settings/ApplicationSettings, each can be checked or not independantly. They should be linked, or I missed something ?

bsongis, 2015-05-26:
This is not the same thing:
1) "Enable automatic backup before flashing firmware" in Settings/Settings/ApplicationSettings = we do EEPROM backups in the directory configured
2) "Backup and restore Models and Settings" in Flash Firmaware panel = we read the EEPROM from the radio, flash the firmware, update the EEPROM and write it to the radio in the updated format. This is useful for radios which don't have automatic conversions

bsongis, 2015-05-26:
This checkbox should be enabled by default.

JivaroFAD, 2015-05-26:
Thanks for the explanation, I understand now the difference :)

bsongis, 2015-05-26:
Would you put again the checkbox enabled by default as it was before? So that we can close the issue ...

JivaroFAD, 2015-05-26:
Done : 88f907abc5b00b36cdf834ad49e66f1d508fe5a6

closed 2015-05-26 by JivaroFAD