opentx/opentx #2343: model get corrupted after upgrading from 2.0.15 to 2.0.99 (from 2015/5/25)

offer-shmuely, 2015-05-25:
model get corrupted after upgrading from 2.0.15 to 2.0.99 (from 2015/5/25)
 
Radio: Tranis plus
old version: 2.0.15 
new version: 2.0.99 from 2015/5/25
model backup by companion 2.0.17
model restore by companion 2.0.99   from 2015/5/25
after version upgrading, the Taranis displayed a message of updating eeprom

Note: how can I send you the eepe files?

The following issues found:
setup page:
Throttle source was CH03 --> changed to THR
switch warning: was different settings --> all changed to upper
failsafe mode: was custom --> disabled

flight mode page:
name changed
fade out: was 0 --> 1.0
trim changed

Inputs page: all have bean lost
curves page: all lost
logical switch page: lost

[image]
[image]
[image]
[image]
[image]
[image]

kilrah, 2015-05-26:
Those result from recent changes, as there are still more changes to be done the upgrade procedure hasn't been updated yet.

Note that the upgrade from 2.0 to 2.1 WILL require major review of models, notably anything related to telemetry will be lost.

bsongis, 2015-05-26:
Thanks, I introduced a shift of one byte ...

closed 2015-05-26 by bsongis

offer-shmuely, 2015-05-26:
Is it ready to be tested it again?

bsongis, 2015-05-26:
I will launch a nightly build right now

bsongis, 2015-05-26:
Nightly build is now launched:
http://jenkins.open-tx.org:8080/job/OpenTX%20Unstable%20-%20Nightly%20builds/

commit ca515a9e (2015-05-26): Fixes #2343