opentx/opentx #2347: Stick Input displayed on graphs

MDMask, 2015-05-30:
It would be ideal if a person could view a graphical representation of stick deflection in the axis they want to monitor / modify.  An example would be an aircraft setup, making knife edge trim adjustments for coupling in roll and yaw.  With a graphical depiction, one could easily see when they achieve a predetermined deflection- and then know how to trim for both roll and yaw.

There are many other examples of its usefulness as well (graphical depiction of throttle curves, DLG modes based stick deflection (e.g cruise mode vs thermal mode with just a single input...with the ability to meter the input), etc....

Thanks for considering!

Mark

kilrah, 2015-05-30:
I must not be understanding correctly what you want, as it would seem to me it's already there (e.g. in inputs). Please detail a bit...

MDMask, 2015-05-30:
Andre-

I just updated to 2.0.17, and see that the Inputs page shows exactly what I was attempting to describe.  I apologize for the additional workload this caused you and the team, but appreciate the help and hard work you have all done to provide this great tool!

 

Best regards,

Mark

closed 2015-05-30 by kilrah