opentx/opentx #2350: x9d crashed

pepper1, 2015-06-01:
hello guys, 
i am new to this so forgive me. i bought a multi from a guy and the tx is a 9xd. worked great till this morning. turned it on and it locked up. had to pull the battery to turn it off. i downloaded companion but i seem to be stuck. i figured my sd card got screwed or something so i have been on the net all day trying to get the tx just to work.... i think i am starting from scratch but im not sure. when i try to load firmware i get the screenshot posted. is there anything you guys can send me to help out? it gets to that point and just stops. 
[image]

bsongis, 2015-06-01:
Does it always stop in the middle of the transfer? It happens here from time to time, but it's always good on the next flash ...

kilrah, 2015-06-01:
First time I see that.

I'd try with another computer first. Do the other flashing methods work and give reliable operation?
If not I'd suspect the processor is somehow shot...

pepper1, 2015-06-01:
it will actually get to different percentages then give that error. it's crazy.  I did actually get it to go to 100%. unplugged and turned on power. nothing, frozen screen. I was using the TX all day saturday. first thing monday... nothing.... I can get to the bootloader and I installed firmware through sd. loaded that, then it says welcome to open TX then freezes/locks up again. every time I have had to pull the battery to turn it off no matter what I do to it.

pepper1, 2015-06-02:
ok..... strange turn of events.... i come home today. try to load the firmware.... 100% bam.... loaded. comes on. turns off. one happy camper. UNTIL!!! i can't get the freaking model to load

when i go to write the model it says to do the 3 finger dance. WHICH i did, usb connected. ugh!!! can you help me load the model? can i just pull the sd card and write it in there somewhere? 

i am guessing i have a bad SD card. which i will be replacing anyway once i get it to work.

pepper1, 2015-06-05:
update. now we're back to not being able to load ANYTHING. along with a box poping up at the bottom saying Nivida GeForce not found or something to that effect. I click anywhere and that little box goes away. 
we are  back to what's in the first post.

bsongis, 2015-06-05:
Would you check on another computer please?

kilrah, 2015-06-05:
> along with a box poping up at the bottom saying Nivida GeForce not found or something to that effect.

Would seem your computer has issues that may or may not cause the problem. Yes try another computer. If that still fails, you might have a hardware issue with your radio.

pepper1, 2015-06-05:
thanks for the help guys. I am able to load firmware by pulling SD , loading into firmware folder, 3 finger salute, load firmware. but nothing on my eeproms. 

I have redone so much,  when you pull up SD card on computer. what folders should be showing? 

also I am using it on an octocopter. is there a general setting of eeproms that I can download for pixhawk?

I bought all of this stuff from another guy so I am having to start over. I have all my settings for pixhawk saved, I am just screwed and newly learning setting up the taranis.

pepper1, 2015-06-05:
oh... forgot to mention my wife's work computer has windows 7. I will try hers this evening when she gets home.

closed 2015-06-12 by bsongis