opentx/opentx #2352: Persistent timer in 9XR

tomekmgr, 2015-06-02:
It would be nice to have a persistent timer in 9XR. 
If there is no eeprom space it may be enabled by compile option.
This option can reduce model memory to 29 (Assuming that will be enough)

kilrah, 2015-06-03:
Not possible because the 9X/9XR don't have software power control to allow saving data on power off.

closed 2015-06-03 by kilrah

xmxwx, 2015-06-04:
Writing to EEPROM frequently increases the risk of its failure. Therefore, to reduce this risk it would be needed to reduce amount of EEPROM writes. There would be a trade-off between the accuracy of such a persistence and the risk of failure. @kilrah proposed to reduce the number of writes until the very last possible moment - the power off. However, that would need a hardware modification. Hardware mods are quite popular for 9X/9XR. If you'd like to propose one which would enable the software to control the power (or maybe instead just provide some extra power on shutdown and notify the CPU that the power is going away in a while) and provide software support for it, it could be integrated in OpenTx.

I have also another idea, also requiring a hardware mod - use NVRAM to store timer's value.

bsongis, 2015-06-05:
We have added a PWRMANAGE in the Mega2560 version of OpenTX, we could introduce this option also in the 9x for those who want to add this mod.