opentx/opentx #2353: Feature request: Special Function to set different failsafe configs

fraternl, 2015-06-03:
It's always difficult to choose a certain failsafe setting as it depends a lot on where the model is at that moment.
I'm flying DLG's and have had some RSSI-warnings whilst flying at 280 metres.

In such a situation I would want the model to start diving and circle down before it reaches an even higher altitude where it might not ever regain a link.
That kind of behaviour is something you don't really want at lower altitudes.

Because there's not that much you can do in the plane itself with the sensors, I thought of having a function to download these particular failsafe settings when you break the 100 metre barrier and then download another failsafe setting for lower altitudes.

fraternl, 2015-09-24:
I never had any feedback on this proposal.
Because it needs an EEPROM change it can only be implemented in 2.2.0
Is there any chance of a dual failsafe setting and a SF to set it?

kilrah, 2015-09-24:
Most likely not.
Heavy in resources, complicated to implement, and... useless if you ask me.

The only way I'd see it happen is if/when a "set failsafe" lua API gets made.

closed 2016-11-27 by fraternl