opentx/opentx #2354: Request, Nativ Mavlink Telemetry in OpenTX

ilihack, 2015-06-03:
Thank you allot for you creat work on the OpenTX firmware fur us an for my Taranis!

but i miss one think, its the nativ mavlink Telemetry compatibility in the comming OpenTX 2.1 like this great work: http://droni.eu/mag/2014/10/12/mavlink-data-displayed-on-taranis-without-converter/

with the next big openTX 2.1 Telemetry update, makes it maybe easy to integrate mavlink due to its flexibility: 
http://openrcforums.com/forum/viewtopic.php?f=45&t=6887&start=90

dk7xe, 2015-09-25:
Would be really a big step forward for all us APM users. And we can get rid of additional HW that is currently needed to have nice Telemetry as it is done with [MAVLink to FrSky S.Port Protocol Converter](https://github.com/Clooney82/MavLink_FrSkySPort)

Gerald

bsongis, 2015-09-25:
Why would it be an OpenTX issue? Is there a Mavlink dedicated module that could be plugged in the module bay of the Taranis?

kilrah, 2015-09-25:
Mavlink telemetry is typically received through a simple transparent serial link, I believe that issue is about integrating native mavlink protocol like there is as an option on the 9x (i.e. in 2.1 a new telemetry handler to decode the stream and populate the generic sensors).

bsongis, 2015-09-25:
But it would be a D-Telemetry, not SPORT, right?

kilrah, 2015-09-25:
No it would be raw mavlink! Like on the 9x. No FrSky gear involved, just receive raw mavlink protocol either on UART3 or sport

Clooney82, 2015-09-28:
Would be a great enhancement for many people.
Best way would if it could be possible to send mavlink data via s.port to RC, but don´t know if this is possible.
Or connect a 433/915 Telemetry Modul direct to RC via serial port and read data from this.

If possible I would prefer a solution without additional hardware.

kilrah, 2015-09-28:
A link on the referenced issue shows that Ardupilot now supports putting out both the hub and the smart port protocol, so I'm not sure adding mavlink support makes much more sense anymore.

Clooney82, 2015-09-28:
Yes ardupilot supports both protocols more or less good, but there are more FCs out there.
For me it would be a nice to have feature which could save weight on smaller copters/helis/planes.

dk7xe, 2015-09-28:
Ardupilot supports currently a few parameters via FrSky protocol as i can see on http://copter.ardupilot.com/wiki/common-frsky-telemetry/
but still not all. 
And as Clooney stated there are other Fc's out there that don't have a FrSky protocol output at all.

The advantage i see with a MavLink parser on opentx is the flexibility we gain on telemetry parameters.
being independent from a implementation (status) on various FC's.

xros, 2015-09-28:
I think doing the telemetry ( in any protocol such as MavLink or others ) in the radio system OpenTX is great.

It can easily let people parse the telemetry parameters. Just as what @dk7xe said, it's more flexible. 

And that will be possible for instance that we can use RC and transmitter as a relay to our ground station software, such as APM2 / Mission Planner using serial ports. Then we do not need the 433/915 3DR telemetry devices anymore.

The full path is like:

GND Station <--> APM
Using:
GND Station <--> USB Serial Ports <--> FrSky RC controller <--> Receiver <--> Telemetry Ports on APM/Pixhawk

How would you think?

kilrah, 2015-09-28:
Using the FrSky link is not possible without FC support, unless the FC gets code to send the mavlink data over smart port (not existing yet, not in our scope, and expected performance would be quite poor due to the FrSky downlink being low bandwidth and mavlink not being optimized for that at all unlike the smart port protocol) an external serial TX/RX set would be needed. And people here have been precisely mentioning they would want to avoid the FC requiring code, so... I see 3 possiblilities:

1. Use additional hardware with no FC-side code with native mavlink on the radio, fed through the serial port
2. Use no additional hardware, but FC needs non-existing code to be written to inefficiently send mavlink data over smart port
3. Use no additional hardware and make use of the already started and more efficient smart port implemensation in AP, just polishing it and adding the missing relevant data

while what people seem to actually want is no hardware nor FC code. 
That's gonna be tough, I'd personally vote foe the 3rd option.

mpjf01, 2015-09-28:
At present thee are two basic sets of solutions for getting Mavlink data onto a Taranis. The first involves a separate piece of hardware such as a Teensy that sits between the telemetry output of the flight controller and the telemetry input of the receiver. It converts the Mavlink telemetry protocol into a FRSKY telemetry protocol that is interpreted as native by OpenTX and can be used in the normal way, most often with .lua scripts.

The second and more recent is the facility to output both FRSKY SPORT and Hub protocols direct from flight controllers, at present the Pixhawk.

Both solutions rely on converting the MAvlink protocol into one or other or both FRSKY protocols. Because the FRSKY protocols do not have many of the data fields that are included in Mavlink, the solutions rely on using "less useful" FRSKY telemetry types and using them to carry quite different Mavlink data. For example, in one solution the FRSKY T1 field (Temperature) has been used to carry the Mavlink Flight Mode data and the FRSKY T2 field to carry the number of satellites. In another project, Ti is used to carry the GPS status and satellite count and T2 used for ARMED status and Mavlink messages,

The different mapping arrangements make using common .lua scripts impossible and extra unnecessary work is needed.

So it would be obviously useful if OpenTX could interpret the native MAvlink protocol without the need for conversion or mapping. I don't think it's just a conversion exercise though as ideally the communication would be two way and there would be a method, such as has been developed already in the Teensy project linked to above, for interpreting and displaying the numerous Mavlink error messages and for keeping the whole arrangement synchronised with ongoing Arducopter/Plane development.

xros, 2015-09-29:
I agree with @mpjf01 , opentx should at least have a systematic of handling native MavLink protocol. 

That whether parsing or sending the native MavLink protocol data depends on what kind of outdoor usages.

In this way, people would focus on Standard Lua Scripts rather than too many different clones.

FC just need to focus on its flight data sending and do not need to focus on what kind of radio that somebody is using.

Let's assume that "What if MavLink Protocol has been changed A LOT(Different versions)"?

Programming in opentx without flashing converter hardware is the key. And yet we don't need to.

And I vote the 3rd option of @kilrah as well. No hardware, handling telemetry in opentx.

kilrah, 2015-09-29:
> And I vote the 3rd option of @kilrah as well. No hardware, handling telemetry in opentx. 

You must have misunderstood me, my option 3 is completing the following:

> he second and more recent is the facility to output both FRSKY SPORT and Hub protocols direct from flight controllers, at present the Pixhawk.

dk7xe, 2015-09-29:
@kilrah 
"while what people seem to actually want is no hardware nor FC code. That's gonna be tough"
may you elaborate a bit more please on the issues you see there.
Thank you!

kilrah, 2015-09-29:
What you said:

> The advantage i see with a MavLink parser on opentx is the flexibility we gain on telemetry parameters. [...] being independent from a implementation (status) on various FC's.

If you want to carry mavlink over FrSky gear then the FrSky protocol needs to be implemented on the FC to be able to talk to the receiver (all the various values don't need to be implemented, but the general communicaiton code still has to be), which you say you'd want to avoid. But if you don't want that then the only solution is separate hardware in the form of tranparent serial modules like what's usually used with mavlink-speaking gear, which @xros wants to avoid.

xros, 2015-09-30:
@kilrah 

> The second and more recent is the facility to output both FRSKY SPORT and Hub protocols direct from flight controllers, at present the Pixhawk.`

Not all users are having a Pixhawk. There are still millions of users using APM2.6 or APM2.8. I know that pixhawk is implanted FrSky protocol, but the firmware for APM was stopped at version 3.2.1 which does not include FrSky protocol.

Handling MavLink in opentx can let people get rid of the hardware MavLink->Frysky converter.  Actually theoretically we do not need a hardware converter.

Is it too troublesome to parsing MavLink protocol data in opentx? 

What programming difficulties would it occur?

kilrah, 2015-09-30:
Read my posts again, I understood that well. But for OpenTX to decode mavlink it has to receive it somehow in the first place.

And for that that there are 2 solutions:
- Carry mavlink over the Frsky link, but for the reason you mentioned that is not appropriate as that would require the frsky protocol and like you said some FCs don't support it.
- Carry mavlink over a set of extra radio modems, but you also said said you didn't want extra hardware.

So my point is that at this point even if we would consider adding native mavlink decoding in OpenTX nobody seems happy with either way of sending the data down.

So decide on that, then we can maybe go forward.

dk7xe, 2015-09-30:
@kilrah thank you for making that clear. At my side the confusion was that i was thinking of the s.port as a transparent port where all the polling for sensors/protocol handling is done by the transmitter but that isn't the case as far as i understand now.
Means we cannot get pure Mavlink over S.Port and the only way via S.Port would be the minimum protocol handling capability on the FC/any other hardware side.

From a standardization standpoint i would still prefer that way cause then we would not misuse FrSky parameters, right?

But this would mean also buy-in from all the dev's currently sending data via S.Port. At the moment i'm  aware of  [ardupilot](https://github.com/diydrones/ardupilot) and [MavLink_FrSkySPort](https://github.com/Clooney82/MavLink_FrSkySPort) doing it that way.

Is my understanding correct?

mpjf01, 2015-09-30:
Just a question. It has been said above and elsewhere that sending the data using the native Mavlink protocol over the FRSKY telemetry link would be too slow (if I read it right). The two methods in use at present therefore convert the Mavlink protocol to Hub or Sport protocol and then send it over the normal FRSKY telemetry link and (presumably) the response is satisfactory. (it is at least with my Teensy setup). As the actual end information seems to be the same, I was wondering why the Mavlink protocol is that much slower?

ilihack, 2015-09-30:
Hello all to together again, I open up the this request, and it's nice to see it's a little bit forward. :-)

I mean and like Kilrah 1 option  methods native Mavlink telemetry integratio in opentx, 

Because I use my FC with openlrsng and on taranis backside the ppm openlrs recivermodule with integrated serial port output for my groundstation  and in don't need an extra telemetry module ☺

with nativ mavlink data parser like for the turnigy rc  we have first: nicer Telemetry vs the mavlink to  frsky telemetry conversion

And we  don't net the teensy board for the taranis  :-)

Maybe it's possible to copy the mavlink parser code from the turnigy to ouer taranis
Greats and sory for my school English.

geofrancis, 2015-12-04:
i am in the same boat. im running openlrsng rc telemetry with pixhawk. so having mavlink status data like i could do with my old modified 9x is essential.

xros, 2015-12-04:
@ilihack @geofrancis Hey there. Same here with a Taranis X9D plus Radio.
We need native MavLink parser in opentx no matter how the data would be carried. :+1:

geofrancis, 2016-07-06:
currently have a taranis sitting useless to me since i have used mavlink on my 9x 9xr and 9xr pro everything i fly is built around it, 

from what i can see is there is 2 ways to do it, 

either read native mavlink
or internally convert mavlink to frsky format.

inis1981, 2016-08-30:
Whats the latest on this?

xros, 2016-09-14:
Lots of guys are using OpenLRS transmitters/receivers which have telemetry enabled. And the baud rate of a OpenLRS transmitter can be set at 4800, 9600, 19200. The mavlink protocol uses baud rate at 9600, 57600, 115200 doc.  Actually parsing native mavlink in OpenTX is not hard... Guys. Really.

(Example of telemetry on OpenLRS software/hardware sending data wrapped in native MAVLink protocol)[http://diydrones.com/profiles/blogs/mavlink-data-displayed-on-taranis-without-converter]

I know that The baud rate of FrSky telemetry is very slow at 1200. Honestly I don't really care this terrible low bandwidth telemetry on FrSky.

Please, we don't have to think only in a Fr-Sky product box. We can build transmitters/boards/system anytime. Many hardware hackers/experts/engineers need this functionality for testing and controlling drones remotely. Years later, we will have/build/buy other radios. OpenTX is the key which lasts forever rather than a unique brand radio product.

bsongis, 2016-09-15:
Don't hesitate to give an hand if it's not that hard, we would really like to include this feature, just not enough volonteers and spare time to do everything while you fly ...

schwabe, 2016-09-15:
If you need an example how to add telemetry look at the Spektrum telemetry commits I recently added (352de898387d00d7d119c603218854dae6fa4cbe and 7c437196646dc56ffe428d97927d4947975811f7)

inis1981, 2016-09-15:
Thanks everyone for working on this, I'm still using gitsly branch however would like to be able to use the latest update. Unfortunately I can't write code.

Snowboard-Nut, 2016-10-07:
Throw my name in the hat for really wanting this also. I use UltimateLRS with Wolfbox Modules in my Initial Retail Release Taranis and I am craving to have the Mavlink data handled in OpenTX without the need for more gadgetry hanging off the back of my Taranis....

gyrex, 2016-10-23:
Hi guys, what's the current status on this project? I'm using a Pixhawk with Ultimate LRS Mini (ULRS) with a Turnigy 9XR Pro and I'm super keen to be able to get telemetry on my transmitter.

kilrah, 2016-10-23:
As you can see it hasn't been touched in a year and a half.

It will likely require a developer with interest in this to join the project and do the coding effort.

gyrex, 2016-10-23:
That's a real shame. I have no idea how to code but couldn't we use/borrow code from other projects who've managed to get mavlink data into 9XR/Pro transmitters?

schwabe, 2016-10-23:
@gyrex With other project you probably refer to er9sky. The new telemetry code in 2.1+ for ARM processors has changed quite a lot and it is not a simple copy&paste. And I think none of current developers has hardware to test that feature.

geofrancis, 2017-01-29:
The best solution I have found so far is using a teensy based mavlink to frsky smart port converter between my openlrsng radio and the taranis. 

The key to getting this to work is the converter is now polling it's own output instead of requiring a frsky receiver to be connected.

The benefits of this us that you can use all the frsky alarms and functions just like a stock frsky system.
If this was integrated into opentx then it would be the best of both worlds.
https://github.com/athertop/MavLink_FrSkySPort

gyrex, 2017-02-27:
Please add this to OpenTx. Manly people need native mavlink support.

3djc, 2017-02-27:
If there is that many people needing it, surely one of them is a developer and can do the required work. We do not refuse it by principle, simply none of us as time or knowledge/interest to do it.

schwabe, 2017-02-27:
As I implemented two extra Telemetry protocols (Flysky and Spektrum), implementing a protocol is not that hard. I even considered implementing Mavlink. However without being able to test it, this is going to be difficult. And for me to be able to test that, I would need a) an APM/Pixhawk and b) something that actually carries Mavlink data over RF, e.g. a 433 Mhz RF module and receiver, which also supports outputting inverted serial data. After that implementating I would have no use for that stuff. And implementing something without having access and remote testing is something which I rather avoid (especially in a spare time project). I think none of the other OpenTX developers has such a setup.

gyrex, 2017-02-27:
Could you implement the solution by using a direct serial link via wires
between the APM/pixhawk and the Taranis? If so, I'm sure some of us could
chip in to buy you a pixhawk clone? They're only ~$60usd these days...

----------------

inis1981, 2017-02-27:
I'm currently setup with a mavlink via 433, however I cant code. Currently the mavlink goes to a BT module and then distributes to a gcs.

mpaperno, 2017-02-27:
When talking of "supporting Mavlink" one needs to define what exactly is to be supported.  Mavlink has hundreds of different messages, many of which are quite specific (eg. to a particular subsystem like mission planning or camera control).  In some cases there are multiple messages which convey basically the same information or otherwise overlap.  It is also easy to make custom Mavlink messages (APM does a lot of that), and in the latest version existing messages can now also be extended with new data fields.   E.g. when using Mavlink in flight controller or GCS code, one usually includes the messages specifically for that FC (which then in turn includes all "common" messages).  

You can see the current "official" varieties here: https://github.com/mavlink/mavlink/tree/master/message_definitions/v1.0 

There is a bit of a misconception that every flight controller/device uses the exact same messages, or that supporting Mavlink means specifically APM/Pixhawk (wrong).  And of course Mavlink is a 2-way protocol, capable of sending commands back to the system -- the discussion here I think has assumed only a receiver capacity, which is fine but should be stated.

So even if there was "someone"  😉  around here who had all the qualifications and equipment, this is still not as straight-forward as it may seem.  (I admit I've only scanned the above discussion so perhaps I missed some finer points.)

I'm not trying to say this is a bad idea, but the request does need to be much more specific than just "please add Mavlink."   

What I do think is a poor idea is adding a specific implementation of Mavlink, eg. for APM.  If that is needed, then there needs to be a way to build OTx firmware with those messages supported specifically (like any other GCS).

inis1981, 2017-02-27:
I'm curious, are we arguing this to argue it or are we arguing it for a practical reason? This discussion is nearly 2yrs old. No specifics have been asked (per your last message i.e. meaning apm mavlink, any specific FC..etc). Just arguments and reasons NOT to implement this. If it can't be done that's one thing, however I feel that is not where you're going with this. 

Now with that said, I have all the equipment to test this, 4 transmitters all setup differently with various versions of opentx, over 10 quads and random FC's. However I can't code. Now with that said, if this can't be done. Say it or close this request. Otherwise, let's put a plan together and make it happen.

kilrah, 2017-02-27:
>  No specifics have been asked

That's the problem! Everything can be done, but not without a proper spec. 

As @mpaperno said there are many variants with little differences that make it a pain to implement and support, e.g. ersky9x has some implementation to display some common Mavlink telemetry items but it's obviously limited to certain uses, and there are scenarios where it breaks regularly. One of the example issues are the different RSSIs reported differentl.

We could just pass raw serial data (or parsed/formatted) to lua and let people do scripts for their own "variants" and FCs. I don't see another way to handle this in the firmware itself without it becoming a maintenance nightmare.

mpaperno, 2017-02-27:
I didn't notice anyone saying that it can't be done (and I don't see why it couldn't).  I think it boils down to effort involved vs. benefits received.  So far with Autoquad I've found it simpler to just use S-Port telemetry since it doesn't require yet another wireless Tx/Rx and can still be implemented "native" on the FC.  With OTx 2.1+ this is now even easier and much more flexible (at the expense of some extra setup needed on radio, of course).  I understand this is not always possible or desirable, but it does beg the question of what exactly people need to see (or hear/feel) on their radio.  I don't see the radio as a viable replacement for a full-featured GCS, but it's great for basic telemetry data and alerts, and in many cases S-Port can replace Mavlink entirely (except for config/setup if the FC relies on it).

hhatlen, 2017-02-27:
I would like to share my (hopefully temporary) solution for displaying Mavlink data on Taranis.
I'm using ULRS 2.0 beta, which gives me a long range bidirectional link that sends RC channels and Mavlink data on the same system.

[http://www.itluxembourg.lu/site/ultimate-lrs-2-0-beta/](url)
[https://www.rcgroups.com/forums/showthread.php?2037442-Ultimate-LRS-40-km-APM-Telemetry-RC-link-for-54-(using-2-OrangeRX-1W-TX)](url)

To be able to see Mavlink data on Taranis I wanted something simpler than the Teensy converter solutions, and after a while I found a post written by Sam Tabor. He made some changes in OpenTX 2.0.12 that made Mavlink data show up in the telemetry screens. The screens show flight mode, arming status, GPS data, heading, altitude, battery status, RSSI and so on. All of the most important things.

[http://diydrones.com/profiles/blogs/mavlink-data-displayed-on-taranis-without-converter](url)

The only hardware change I had to do was to invert the serial levels between the TX module and Taranis.

[https://www.rcgroups.com/forums/showpost.php?p=26953006&postcount=2802](url)

The disadvantages of using this solution is that you're basically stuck with OpenTX 2.0.12, and you can't use it for scripts, alerts and so on. But in my case there are times when I don't want to drag along a full GCS out in the field, so being able to see Mavlink data on the Taranis is a huge advantage.

I cross my fingers and hope that Mavlink will get implemented, at least with the most important things that I mentioned above.

mpaperno, 2017-02-27:
The current Mavlink implementation in OTx seems to center around one flight controller, APM.  This is not really Mavlink, this is "APM telemetry."  It may happen to be compatible with other devices, or it may not.  This kind implementation doesn't interest me personally.

What I'm picturing would read incoming Mavlink messages and (essentially) turn the data into virtual sensors.  Some of the data would probably be usable on "standard" screens (numbers/bars), and for others it would be up to Lua script authors to parse/display (e.g. flight mode name).

So, what I/we would need is a list of Mavlink messages to be supported.  This should be as small and as generic as possible/reasonable.  They should all be from the [common](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml) definitions.

mpaperno, 2017-02-27:
I'll start the list (maybe move it to a better place later).  
Items with a `?` are ideas/proposals and hopefully some of the duplicates at least are not actually needed.  
Items with `*` are [currently supported](https://github.com/opentx/opentx/blob/next/radio/src/telemetry/mavlink.cpp#L360) in OTx implementation (though not necessarily all data values are actually used).  
Add suggestions in your comments.

MAVLink messages to support:
1. HEARTBEAT `*`
1. SYS_STATUS `*`
1. STATUSTEXT `*`
1. RC_CHANNELS_RAW `*`
1. GPS_RAW_INT `*` `?`
1. GLOBAL_POSITION_INT `?`
1. GPS_STATUS `?`
1. BATTERY_STATUS `?`
1. SCALED_IMU `?`
1. SCALED_PRESSURE `?`
1. ATTITUDE `?`
1. RC_CHANNELS_SCALED `?`
1. RC_CHANNELS `?`
1. RADIO_STATUS `*` `?`
1. MISSION_CURRENT `?`
1. NAV_CONTROLLER_OUTPUT `*` `?`
1. VFR_HUD `*` `?`

hhatlen, 2017-02-27:
That looks good!

I only have these two suggestions in addition:

18. ALTITUDE `?`
19. MISSION_ITEM_REACHED `?`

ALTITUDE, because of altitude_terrain.
I'm not quite sure how MISSION_CURRENT and MISSION_ITEM_REACHED behaves at the end of a mission, so I just added it here.

gyrex, 2017-02-28:
>    1. ALTITUDE ?
>    2. MISSION_ITEM_REACHED ?

Is anyone keen to set a bounty to accelerate the implementation? Id be
happy to chip in...

----------------

Chadwide, 2017-03-02:
Just wanted to add that this is important to me also. It's preventing me from upgrading my 9x to a Taranis. Also happy to chip in to a bounty.

hsteinhaus, 2017-12-20:
+1 for a raw MAVLink telemetry link in OpenTX. This is especially useful for non-FrSky radio links. Additionally, this choice usually offers access to more telemetry data than any bridging solution.

I just did a bit of code reading. If I understand the current situation correctly, almost anything requested in this issue and more is already there (radio/src/telemetry/mavlink.cpp). However, in contrast to e.g. CrossFire/CRSF, there is no means to select this type of telemetry/RF module mode in a per model choice. Instead, you have to compile a special firmware binary with "-D with MAVLink enabled and lose any other kind of telemetry.

@bsongis @schwabe @mpaperno 
Is there any particular reason for the choice to implement MAVLink telemetry as a compile time option (e.g. memory/other resources)? Do you think it would be viable to change that?

kilrah, 2017-12-20:
The compile option is a relic from the support as it was back in the 9x days where you could not handle more than one protocol at a time for resource reasons.

It was never ported to the new more powerful radios.

hsteinhaus, 2017-12-21:
Ok, that's what I expected. Is there a reasonable chance to get a PR merged that adds an "external RF mode" or two? 

I am considering to add two additional choices for the external RF mode: 
 1) MAVLink telemetry + PPM pulses
 2) MAVLink telemetry + MAVLink control

The old code also seems to contain some vehicle parameter handling. While this part contains most of the complexity (as this is FC-specific and affected by chances in the FC code), I do not really see any reasonable use case for that in the age of smartphone and tablet GCS.

schwabe, 2017-12-21:
If you do the work and it fits there is a good chance for that be merged. Keep in my that the telemetry pin is always inverted (by HW design). For MAVLink telemetry alone you don't need an extra RF module mode. When you select PPM pulses you can select the telemetry type under telemetry. 

You might take a look at the other telemetry implementation (multi, flysky and spektrum) too, they are much simpler and easier structured than Smartport

The old Mavlink code is not really fit for the job as it is hardcoded for the old telemetry system, starting fresh is probably a better choice.

kilrah, 2017-12-21:
The annoyance with mavlink (and one of the reasons nobody bothered yet) is the diversity of combinations people want to use.

If you do an implementation it will need to allow mavlink telemetry with any of the existing internal and external control choices, and include support for dual telemetry sources (expecting mavlink would always come from the additional serial port, and be able to be received simultaneously as the selected control method's telemetry).

mpjf01, 2017-12-22:
As has been stated there are apparently hundreds of possible diagnostic/error messages sent by the FC using Mavlink to the GCS for operator attention. Is the plan to deal with these?

Mission Planner allows the user to show on the Flight Data screen a selection of the 215 items of data that can be sent from the FC, this is in addition to the messages. Will these be supported? Are all the messages determined in the GCS from the values of these data items or are some or all of the messages worked out on the FC and sent as discrete data?

mpaperno, 2017-12-22:
@mpjf01 Please at least read comments starting from [here](https://github.com/opentx/opentx/issues/2354#issuecomment-282792296), and see the list of messages I started below that.  Obviously a subset will need to be chosen.

mpaperno, 2017-12-22:
I still don't really see how it would be more useful than just S.PORT from the FC directly... how much data can one really use on a radio?  

I also get the impression that only APM users are interested in this, but perhaps I'm wrong.

> This is especially useful for non-FrSky radio links.

Such as...?  Do they have native telemetry?  Just curious.

One simple solution one could do right now is simply convert the Mavlink data to S.PORT with a Teensy or whatever.  It's relatively easy to do (lots of projects out there for it), is easy to customize (e.g. to support the messages of interest), and it keeps code out of OTX which is already quite complex,  I realize the DIY approach isn't for everyone, but with a bit energy it could be made into an almost plug/play thing (at least as much as OTX is). 

> Additionally, this choice usually offers access to more telemetry data than any bridging solution.

True, but on OTX side support needs to be added on a message-by-message basis, so it's still a "what does it support" scenario.

Not to say I'm against Mavlink in OTX, I just still wonder if it's worth the effort (not just initially but also support/maintenance).

mpjf01, 2017-12-22:
@mpaperno I had already read the whole thread (again) including the noted sections. They don't deal with my question as I understand it. I see a possible distinction between the Mavlink data items - things like satellite counts, flight mode, GPS status and so and, and the error messages presented to the end user - things like "AHRS ERROR", ARMED, DISARMED and so on. I appreciate that a message could theoretically be a direct "decode"of the data or it might be derived from a calculation performed on one or more items of data - just not sure if the second is the case or not, and if it is, where the calculation os performed - in the FC or the GCS.

In any case, all of the data items and messages have been requested or have been found to be necessary by someone in the past. I would therefore question both the ability of anyone to come up with an agreed subset and the wisdom of doing so unless there is some method worked out to avoid the situation where the end user incorrectly interprets the absence of an error message as being the absence of an error, keeps flying when he shouldn't and experiences unwanted consequences.

hsteinhaus, 2017-12-22:
>As has been stated there are apparently hundreds of possible diagnostic/error messages sent by the FC using Mavlink to the GCS for operator attention. Is the plan to deal with these?
At least with APM, the messages are just STATUSTEXT, so one message only, plus text as data. In contrast to these S.PORT tunneling solutions, that cannot cope with text on the link, I do not see a reason for parsing them. BTW, this is one of the most important reasons why I want to get rid of MAVLink-to-anything bridging in the telemetry data path.

> Such as...? Do they have native telemetry? Just curious.
I use Crossfire, it obviously has telemetry and it is well supported with OpenTX. Unfortunately, as with any bridging solution, some very interesting data, that are contained in the original data stream, are unaccessible (text messages, vibration info, raw servo output - so basically all information related to vehicle health).

> True, but on OTX side support needs to be added on a message-by-message basis, so it's still a "what does it support" scenario.
For me, LUA access would be all I need (I am running custom telemetry screens anyway). So there is no need to present any of the 200+ fields in the GUI. However, I would would definitely try to fill existing telemetry fields for ease of use.

[20 more replies]

closed 2019-11-15 by stale[bot]