opentx/opentx #2355: Internal and external XJT not work together.

janekx, 2015-06-03:
I was ask by the customer for help, that when he use on Taranis internal XJT D16 with X8R for telemetry and external XJT LR12 with L9R together, the internal XJT losing telemetry and RSSI skip much.  

I was sucessful with reproduction. 

internal XJT paired with X8R on D16 mode
external XJT only powered (no need RX to work) and switched to D16 or to LR12 
the internal XJT start loosing the signal and telemetry, RSSI skips from 80 to 40 or 0 again and again

OpenTX 2.0.17 and 2.0.15 same issue
XJT firmwares is non EU so 2014 year. 
Taranis+

kilrah, 2015-06-04:
You must disable telemetry on the external module by  putting both dip switches up.

closed 2015-06-04 by kilrah

janekx, 2015-06-04:
This setting is in manual described like - not defined. Will try thanks

janekx, 2015-06-04:
Tested, this is not the right solution, please reopen, 
When Internal module lost connection, the RX not show lost signal (still green) but telemetry not work.

kilrah, 2015-06-04:
You need to rebind the external module after changing the switches.

janekx, 2015-06-04:
It happend even no RX is connected to external module, only powering the external module start this behaviour.

kilrah, 2015-06-04:
Again you need to put both switches up AND BIND a receiver. Until then the external module's telemetry will not be disabled.

kh4, 2015-06-04:
One option is to cut off the sport pin in the module bay (lowest pin), This ofc. is a bit of a permanent mod.

janekx, 2015-06-04:
Taranis off
XJT external both switch to ON (UP - not defined in manual)
Turn on Taranis
bind internal in D16 mode 5 with X8R 
telemetry OK 
power on the external set XJT and LR12 
telemetry lost
bind L9R, L9R comes green after a bind process
Telenetry not work, RSSI 0 
Turn Taranis off 
Turn Taranis On 
X8R is green, L9R is green, Rssi drops to 0 and telemetry lost, X8R still green and L9R  still green.

kilrah, 2015-06-05:
Hmm I checked the old discussions and it seems the 2 DIPs up might only work when the module is fed with a PPM signal, and would be "D16 without telemetry". That's not what I'd have wanted when I told FrSky what to do about this, but it seems it went unnoticed.

I don't have equipment to do tests where I am now, but if that's correct then you can't have LR12 with telemetry disabled, the only solution is indeed to isolate the lower pin of the external module (maybe better to open the module and cut the trace or remove an inline resistor if there's one than cut the radio's pin of course)

janekx, 2015-06-05:
Hmm this is not a good solution for customers, I think it can be easily to work in opentx, some telemetry source switch right below internal and external module setup.

kilrah, 2015-06-05:
Nothing can be done in OpenTX. There is only one hardware telemetry line that goes to both modules, and obviously only one can talk on it at a time.

The only solution is changing module behavior and that's not in our control.
BUT again it *should* be possible, I remember testing back in the days. Maybe the feature got removed by FrSky in recent module firmware, I don't know.
If I get time and find an XJT I'll try to check.

janekx, 2015-06-05:
I see, I can check it too. I will ask FrSky about FW for XJT with both swich up. 
Thanks 
Jan

dgrat, 2015-08-01:
I seem to have the same problem with a D8 receiver,
and I don't know what is the reason.

geofrancis, 2016-04-28:
Was this ever resolved? 

I am having the same problem with my taranis, I need to use both internal and external Xjt. And telemetry flickers off and on as soon as both modules are powered.

Is the only solution still avaliable to isolate the smart port pin from the external module?

kilrah, 2016-04-29:
Set both DIP switches of the XJT to ON  and rebind, that will disable its telemetry output.

kevinlynn, 2016-08-23:
kilrah: I tried your solution and it still doesn't stop the beeping on my xjt w/ 9x+smartieparts.

kilrah, 2016-08-23:
Then you're most likely not the same case as OP. Can't help without details about your setup, and that should be discussed on the forum.

geofrancis, 2016-08-29:
both dips up then rebind the external module worked fine for me using 32channels via 2x x8r's