opentx/opentx #2357: Makefile variables / #defines clean-up: BOLD, EXTSTD

xmxwx, 2015-06-08:
Why would anyone need a build without defined(BOLD)?

Why is defined(EXTSTD) used to decide whether to include BOLD_SPECIFIC_FONT?

Why the condition for applying some minor LCD printing changes is different from conditions for BOLD_SPECIFIC_FONT?

    fonts.h:#if defined(BOLD_FONT) && (!defined(CPUM64) || defined(EXTSTD)) && !defined(BOOT)
    gui/9X/lcd.cpp:#if defined(BOLD_FONT) && !defined(CPUM64) || defined(EXTSTD)
    gui/Taranis/lcd.cpp:#if defined(BOLD_FONT) && !defined(CPUM64) || defined(EXTSTD)

bsongis, 2015-06-09:
1) Because some people don't like this BOLD font!

2) and 3) Because EXTSTD costs far less flash than all others EXT...

kilrah, 2015-06-09:
Basically:
- On M64 with FrSky telemetry we're short on flash so BOLD is only included if the user specifically chooses it with the option
- On M64 without FrSky telemetry there are no flash problems so it can be included by default. 
- On non-M64 there are no flash problems either so it can also be included by default. 
- No bold font in the Taranis bootloader as we're short on flash and it isn't used anyway.

kilrah, 2015-06-09:
Hmm actually only BOLD_FONT would be needed no? And in companion BOLD is available for M64/FrSky...

mhotar, 2015-06-09:
> Why is defined(EXTSTD) used to decide whether to include BOLD_SPECIFIC_FONT?
Three is alternative method to create BOLD text without predefined/included font data. It writes standard text twice over itself shifted by one pixel
 It is not perfect, but uses less flash. It is used for M64/Frsky instead of the proper Bold font.

closed 2019-08-19 by kilrah