opentx/opentx #2358: intermitent boot issue with T9XR Pro

blairfancy, 2015-06-10:
Hi,
just to report that sometime T9XR pro take very long time to boot. I have seen this issue on two radio.
FW 2.0.17.
 Is there any log i can post to help on this issue?

projectkk2glider, 2015-06-16:
Please provide more data such as:
* does presence of SD card influence this
* can you point out which part of boot sequence takes a long time

blairfancy, 2015-06-16:
I don't know the full sequence of boot but it long time occurs when logo appear and it can stay around 1min.
I will try without SD to check that.

projectkk2glider, 2015-06-17:
Also, when delay happens, try pressing some keys to see if it influences anything.

blairfancy, 2015-06-17:
For keys i ve already tried and nothing append and as far i remember it is non constant delay.

LapinFou, 2015-06-23:
Do you use some permanent timers?
Does the long boot-up delay depend of the selected model?

blairfancy, 2015-06-26:
Hi after some time spend with my 9Xr pro issue is related to SD card. Without SD card bug does not occurred. Actually voice is activated perhaps it is what that freeze radio on bootup.

projectkk2glider, 2015-06-26:
Now that you figured out that the issue is related to SD card we need to find out if:
* the issue is caused by **your** SD card or radio combination - that means that it only affects you
* the issue is caused by several SD cards - that means that it affects several/all users.

What do you think, which one is the case here?

blairfancy, 2015-06-26:
I m sure that i m not the only one affected by this bug. I see it on another radio. SD card is delivered perhaps it low cost card. I ll try with another one.

Tank86, 2015-07-02:
Hi, Same issue here, radio freeze at power-on when sd card (4gb stock sd) is plugged, no issue without sd.
Seen issue into 2 different 9xr pro.
Issue didn't occurs on stock ersky9x firmware.

projectkk2glider, 2015-07-02:
Can you both tell us which was the last version of OpenTX that did not have this problem?

Tank86, 2015-07-02:
I am new at RC so It's the first version of OpenTx I use ...
Is there is some data to log to help to find the issue ?

projectkk2glider, 2015-07-02:
@Tank86 no log in this case. Maybe @blairfancy used some previous version that worked. Lets wait for his answer.

blairfancy, 2015-07-02:
Unfortunately i can t remember if older FW had same issue but with sandisk 2g sd card error was gone. Perhaps some delay are required to lets slow sd card initializing properly

Tank86, 2015-07-03:
My only other micro sd (8gb) is not recognized by the radio ;(.
Additional information, the freeze occurs just before the sound "Welcome to OpenTx", and when the radio unfreeze, the sound is played.

bsongis, 2015-10-20:
Duplicate of #2852 
Nightly builds are available, would it be possible to have some tests before we release?

closed 2015-10-20 by bsongis

blairfancy, 2015-10-21:
Sure i can test it could provide me link to companion 2.1.4 for ubuntu?

projectkk2glider, 2015-10-22:
http://downloads-21.open-tx.org/nightly/companion/

blairfancy, 2015-10-22:
Is it possible to download trunk version of opentx FW with conpanion 2.1.3?

kilrah, 2015-10-22:
What do you mean by "trunk version"? Current master status?
If that's what you mean you can download firmware nightlies once you've installed a companion nightly and set in the preferences to use the nightly builds.

blairfancy, 2015-10-22:
Hi v installed nightly build but cant find preference to use nightly FW

kilrah, 2015-10-23:
> note that it is compiled companion from git not nightly build

Then you won't have it by default. For that you need to manually configure cmake to define ALLOW_NIGHTLY_BUILDS with
```
cmake . -DALLOW_NIGHTLY_BUILDS
```
in the build directory.

projectkk2glider, 2015-10-23:
And make sure you start from the clean directory when changing cmake options, otherwise (in my experience) they do not take effect.

kilrah, 2015-10-23:
Not here, I commonly add/remove defines that way when testing and never had any issue. The command above precisely should actually be used on an existing build dir.

blairfancy, 2015-10-23:
correct syntax is:
<code> cmake -D ALLOW_NIGHTLY_BUILDS=true ../src </code>
finally i can flash radio without problem,  model setup also properly transfered.
Not reproduced long boot issue even with my old sdcard. I ll try to play more this week end.

blairfancy, 2015-10-24:
Ok after several try bug occurred then this version does not solve the issue :( .

blairfancy, 2015-10-24:
Is it possible to implement initialization debug like linux boot?
ie:
loading bootloader ---> Pass
loading opentx kernel --> Pass
Initialization of LCD --> Pass
Display logo --> Pass
Activate welcome sound --Pass

regards

projectkk2glider, 2015-10-24:
I suggest you continue in the issue #2852, this issue is closed an marked as a duplicate of #2852.