opentx/opentx #2360: Simulator 9xr-pro not working in companion 2.0.99

procha170, 2015-06-13:
Data from companion are not transferred into simulator . Buttons not working except + an - .In older builds buttons worked so I tested that simulator works normally and no setting from companion is transferred.I tested sever version of companion 2.0.99 and it's always the same.

bsongis, 2015-06-28:
Not only a SIMU bug

procha170, 2015-06-29:
Hello,

 

I don’t understand if the reporter problem was solved. I tested last build and it look better – all buttons work in SIM but after setting some model in companion and pressed simulate button there is no setting transferred into SIM (all mixes are empty in SIM)  . At this moment I’m able work with simulator like with real radio and if I set mixes here it works.

 

Ales

kilrah, 2015-06-29:
What "last" build? Today's should be OK, while yesterday's probably wasn't.

bsongis, 2015-06-29:
Almost sure it's a Windows only bug, in MixData I mixed uint32 and uint16 and VC++ doesn't like that ...

kilrah, 2015-06-29:
I am on Windows and can't see an issue, which is why I asked...

kilrah, 2015-06-29:
Woops, 9XR-Pro... forget that, I missed that bit. Broken for me too then.

commit 18d8edce (2015-06-28): Fixes #2360
commit 8b9265a3 (2015-06-29): Fixes #2360
closed 2015-06-29 by bsongis