opentx/opentx #2362: INPUTS - Weight param unnecessarily limited to +/-100

jmrware, 2015-06-13:
Summary
------------
The WEIGHT parameter when editing an INPUT line is unnecessarily limited to +/-100.

Discussion
--------------
On the MIXER page, a single mix line can be set up where the output goes from -100 to +100 as the stick input goes from -100 to 0 (zero), then the output remains at +100 as the stick moves from 0 to +100. (This is used as a "NORMAL MODE" throttle curve for a CP heli.)

Here are the x,y coordinates for the equivalent three point curve:
Input Output
-100  -100
0  +100
+100  +100

With a MIXER line, this is easily achieved using either DIFF (or EXPO) with a WEIGHT = +200 and OFFSET = +100. (Note that this same function is useful for logically converting a 3-position switch into a 2-position switch)

However, in the INPUTs screen, (which has a similar selection of functions and parameters to work with), the maximum value for the WEIGHT parameter is limited to +/--100. Not sure why that limit is enforced, because its the output of the INPUT which must be limited to +/-100 (the WEIGHT param can be >100). Seems to me that this limit serves no useful function.

The current work-around for this shortcoming is to simply use a MIXER line rather than an INPUT (or explicity use a custom curve). The overall model setup would be a lot cleaner if this could be done in the INPUTs screen.

hrenz, 2015-06-14:
And:
Trim at Input with negativ weight (-100)  don't change direction
Trim at Mixer with negativ weight (-100) change direction

projectkk2glider, 2015-06-16:
We should be careful here, we already have a problem if input is outside the +=100% range #2284

closed 2020-07-07 by stale[bot]