opentx/opentx #2364: Taranis Plus S2 pot problem

PoriG, 2015-06-16:
Hi guys!

I have the latest revision Taranis Plus with the latest software from FrSky. When I try to set my S2 pot as an input source by turning it while the source option is highlighted, the transmitter sometimes detects it as S3. This mostly happens close to the center point of S2. In my settings S2 is set as pot with detent and S3 as none. Also calibrated the pot multiple times. Here is a video someone posted on youtube about this behaviour: http://youtu.be/NeWSy73GsEs
Anyone has any idea why this could be happening? Otherwise the pot works fine when I select it as source manually. S3 is not installed in my radio.

Thanks for any info!

projectkk2glider, 2015-06-17:
@PoriG are you willing to test the fix for this?

If yes, please backup your radio (models and firmware) and then flash this file http://jenkins.open-tx.org/nightly-20/2015_06_17/opentx-taranis-plus.bin

The flash file posted above has some default options which might be different from your selection. Ignore that and focus on S3 issue and report back.

We also need somebody else with Taranis Plus that has installed S3 pot to test if it still working.

PoriG, 2015-06-17:
Sure! I will test it this afternoon. I will report back.

PR #2365 (2015-06-16): Fixes #2364: use fixed analog value for non-connected pots
**WARNING** this is completely untested! I have no radio with S3 pot.

Closes #2364 

If accepted I will port it to the `next`
commit b5cf6a13 (2015-06-16): Fixes #2364: use fixed analog value for non-connected pots
commit f32ff084 (2015-06-16): Merge pull request #2365 from opentx/projectkk2glider/issue_2364_s3_ghost_readings

Fixes #2364: use fixed analog value for non-connected pots
PR #2370 (2015-06-17): Fixes #2364: use fixed analog value for non-connected pots (ported fr…
…om master)

Closes #2364 

Again this was **NOT TESTED**
closed 2015-06-17 by bsongis