opentx/opentx #2367: Extended limits not observed when subtrim mode = "^"

RC-SOAR, 2015-06-16:
Using OpenTx 2.0.15 on Taranis

1. Enable Extended Limits.
2. Set up two aileron channels, one with subtrim mode "=", and the other "^"

    CH1 - mode = "="     subtrim=50     min=-100     max=100
    CH2 - mode = "^"     subtrim=50     min=-50       max=150

I would expect these to be functionally identical. However CH1 clips at 100%, CH2 clips at 150%.

Same on both Taranis and OpenTx Companion.

Here's a [minimal .EEPE file](https://drive.google.com/file/d/0B7eJi6dZb_Ukdkh6b3ZPYWlSWTQ/view?pli=1) demonstrating the issue.

JivaroFAD, 2015-06-16:
It's normal result : extended limits allows to set a limit above the classical +/-150%, but it does not change the limit. If you set +100%, no order can exceeds +100%.
The subtrim mode change only the lineraty of response with trum (shift input or ouput in mixer). In all the case, you have to keep in mind that the limit is the maximum travel rate, not a simple weight for servo.

kilrah, 2015-06-18:
> However CH1 clips at 100%, CH2 clips at 150%.

No, CH1 clips at 100, while CH2 reaches 150 without clipping. That's the difference between the 2.

Works as designed.

closed 2015-06-18 by kilrah