opentx/opentx #2371: precision for telemetrie voice

diablol, 2015-06-17:
opentx 2.1

e.g. my voice call "play value" for V_min only says "3V". But we need at least prec1 or actually prec2. 
(even able to choose?)

bsongis, 2015-06-17:
Which telemetry field are you playing? Is it the same for field, field- and field+? What is the PREC you choosed for this telemetry value. Here I use Cells, it's ok with PREC1. PREC2 normally won't be accepted.

diablol, 2015-06-18:
yes cells works.
but most others have wrong voice units. C° is watts mph is watts and a few others are wrong too.

I refered to a custom sensor for  lowest cell voltage. 
field+- is the same

bsongis, 2015-06-18:
Did you install 2.1 sounds?

diablol, 2015-06-18:
ehm nop.

where to get?

bsongis, 2015-06-18:
http://www.open-tx.org/links.html

diablol, 2015-06-18:
ok I didnt knew that.

now she says "3.9" without "volts". Also for "cels" the unit is missing too.

AND all units are shifted one element in the unit list. So if the list is e.g. : volt,amps,mah,mph,knots,celsius,fahrenheit.

amps are called mah and knots fahrenheit etc.

that may also is the reason for voltage having no unit..

bsongis, 2015-06-18:
It means you don't have the latest nightly builds, I fixed this very recently

diablol, 2015-06-18:
It means I am quiet now ;)

bsongis, 2015-06-18:
I start a new 2.1 prerelease now, please confirm we can close this issue

diablol, 2015-06-19:
installed 2.1 sound and recent nightly, but unfortunately units still are shifted

bsongis, 2015-06-19:
"volts" are good, right? I only tested volts myself

diablol, 2015-06-19:
on my taranis not really. for cells unit is missing. amps are volts.

commit 6a3f277f (2015-06-19): Fixes #2371 (french sounds shifted) - Now only en / fr TTS sounds are
correct, I need help for all other languages!
closed 2015-06-19 by bsongis